aircrack-ng/rtl8812au #717: Error kernel 5.4.61-v7l+ raspberry pi 4

sagaardu, 2020-09-07:
hello I have updated my raspberry and it has stopped recognizing the card.

I have uninstalled the driver but when I reinstall it it throws me the following error

```
~/rtl8812au $ sudo ./dkms-install.sh
About to run dkms install steps...
Error! DKMS tree already contains: rtl8812au-5.6.4.2
You cannot add the same module/version combo more than once.
Error! echo
Your kernel headers for kernel 5.4.61-v7l+ cannot be found at
/lib/modules/5.4.61-v7l+/build or /lib/modules/5.4.61-v7l+/source.
Error! echo
Your kernel headers for kernel 5.4.61-v7l+ cannot be found at
/lib/modules/5.4.61-v7l+/build or /lib/modules/5.4.61-v7l+/source.
Finished running dkms install steps.
```

and make:

`:~/rtl8812au $ sudo make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.4.61-v7l+/build M=/home/pi/rtl8812au  modules
make[1]: *** /lib/modules/5.4.61-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:2245: modules] Error 2
`

Crono141, 2020-09-09:
Same issue, only with raspberry pi 3 and kernel 4.14.98-v7+

kogapf, 2020-09-11:
RTFM, but anyway:
`sudo apt-get install raspberrypi-kernel-headers` for the headers, then
`sudo ./dkms-remove.sh` to clean the system,
then make sure that the Makefile is patched accordingly (see README), finally
`sudo ./dkms-install.sh`

sagaardu, 2020-09-12:
Hi, sorry, but not work

```
@raspberrypi:~ $ sudo apt-get install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20200819-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

------------------------------------------------------------------------------------

raspberrypi:~/rtl8812au $ sudo ./dkms-remove.sh
About to run dkms removal steps...

------------------------------
Deleting module version: 5.6.4.2
completely from the DKMS tree.
------------------------------
Done.
Finished running dkms removal steps.
---------------------------------------------------------------------------

raspberrypi:~/rtl8812au $ sudo ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.4.61-v7l+ cannot be found at
/lib/modules/5.4.61-v7l+/build or /lib/modules/5.4.61-v7l+/source.
Error! echo
Your kernel headers for kernel 5.4.61-v7l+ cannot be found at
/lib/modules/5.4.61-v7l+/build or /lib/modules/5.4.61-v7l+/source.
Finished running dkms install steps.

```

kogapf, 2020-09-12:
Try `sudo apt install --reinstall raspberrypi-kernel raspberrypi-kernel-headers`, reboot, and do the steps above again.

sagaardu, 2020-09-14:
perfect!!! it works!!
thanks

Nodi-Rubrum, 2020-09-15:
Trying to use make and make install, still get -mgeneral-regs-only error even following steps in readme.  Should parameter by only 'arm' if 64 bit Pi3 or Pi4?  Should it be something like arm7 or arm64?

pansila, 2020-10-30:
@Nodi-Rubrum Have you fixed it? ‘-mgeneral-regs-only’ error

Nodi-Rubrum, 2020-10-30:
Sorry, been buried.  Will try this weekend.

bakwc, 2021-09-24:
`sudo apt install --reinstall raspberrypi-kernel raspberrypi-kernel-headers` helped