aircrack-ng/rtl8812au #720: any chance for new updates for windows?

0lm, 2020-09-16:
hi all
the chance might be very low, but i wanted to ask anyways.
i have a AWUS036ACH by Alfa Network and it exactly used the rtl8812au drivers. and on linux, it works perfectly. my problem is windows 10. when i install the wpa3 drivers offered by alfa networks, occassionally i get the Evnt ID 11006 error a few times a day. the wpa2 drivers on the other hand are stable. but i would like to be able to use wpa3. but yeah, i suffer these reconnect issues. it is not because of my router. that error only occurs on windows 10 when wpa3 drivers are installed. i have no idea why and how to solve it. my connection is always stable but these occassional short disconnects labled as ID 11006 are annoying.
anyone has experience with that?

nomadmzl, 2020-09-17:
The latest drivers are [here](https://files.alfa.com.tw/?dir=%5B1%5D%20WiFi%20USB%20adapter/AWUS036ACH)

0lm, 2020-09-18:
> 
> 
> The latest drivers are [here](https://files.alfa.com.tw/?dir=%5B1%5D%20WiFi%20USB%20adapter/AWUS036ACH)

Thanks for the answer. I already had the latest drivers installed. Thats the reason why I tried here. I cant seem to solve the problem and there is no way to find out where the roots of this errors are.

nomadmzl, 2020-09-18:
Have you looked at the [reason codes](https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-server-2008-R2-and-2008/cc727757(v=ws.10)?redirectedfrom=MSDN) for that eventID?

0lm, 2020-09-18:
> 
> 
> Have you looked at the [reason codes](https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-server-2008-R2-and-2008/cc727757(v=ws.10)?redirectedfrom=MSDN) for that eventID?

I actually did, but my reason codes do not exist there. for ID 11006 it is 0x48005. By now the error messages changed after Windows cumultative update a few days ago. now i actually ID 11004, (wlan security stopped), before the error event id 8002 appears with the reason code 229378, which is also not existing on the event id 8002 page of microsoft.

kimocoder, 2020-10-09:
We dont to Windows here.. sorry

closed 2020-10-09 by kimocoder