aircrack-ng/rtl8812au #721: rtl8811au on Raspberry Pi4: driver module build and loads, but no interface device appears

TheOneWhoseUsernameWasTaken, 2020-09-17:
Hello!

I'm trying to get a Logilink WL0238 adapter (rtl8811au) to work on a RPi4. I built the module according the instructions, with one seemingly necessary deviation: The instructions say that for the Pi4, I need to to build the ARM64 module. But I'm using the normal 32bit Raspbian (now Pi OS), so I went with the Pi2 sed-command.

I also disabled the built-in wifi by blacklisting brcmfmac and brcmutil. The reason I want to use an USB adapter is that I had reliability issues with those, so I wanted to test a different adapter with a real antenna.

I ran the dkms-install.sh command, it completed without errors, and after a reboot, I could load the module. But no wifi interface device seems to show up. 'iw dev' has no output, the desktop GUI says it can't detect any wifi devices.

What am I doing wrong?

```
# uname -a
Linux cammon 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

# cat /etc/debian_version 
10.4

# modprobe 88XXau

# lsmod | grep 88XXau
88XXau               2027520  0
cfg80211              679936  1 88XXau

# dmesg | tail -n4
[   32.537570] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   32.578954] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   32.646534] 88XXau: loading out-of-tree module taints kernel.
[   32.679229] usbcore: registered new interface driver rtl88XXau

# iw dev

# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:42:30:ac brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.77/24 brd 192.168.0.255 scope global eth0
       valid_lft forever preferred_lft forever

# cat /proc/net/wireless 
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22

# ls /sys/class/net/
eth0  lo

# lsusb -D /dev/bus/usb/001/003 
Device: ID 0bda:c811 Realtek Semiconductor Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xc811 
  bcdDevice            2.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11ac NIC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

ulli-kroll, 2020-09-21:
Next time you should use/add the USB id in the title.

`0bda:c811`
is part of the vanilla rtl88cu device driver from Realtek
either use
https://github.com/ulli-kroll/rtl8821cu
or this branch
https://github.com/ulli-kroll/rtw88-usb/tree/development/chipset/rtl8822cu

TheOneWhoseUsernameWasTaken, 2020-09-21:
Well, the [manufacturer's own data sheet](http://logilink.org/media/datasheets/WL0238.pdf) states as chipset: "RTL8811AU". Maybe it was naive of me to just believe that.

None of the drivers you linked to will work for me. First one doesn't even build:

```
# LC_ALL=C make
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.4.51-v7l+/build M=/root/rtl8821cu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.51-v7l+'
Makefile:648: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.51-v7l+'
make: *** [Makefile:480: modules] Error 2
```

And the second's build process doesn't match the instructions, and the final modules have unknown symbols. See:
https://github.com/ulli-kroll/rtw88-usb/issues/10

satmandu, 2020-09-28:
@TheOneWhoseUsernameWasTaken If you have a arm7l userland with an arm64 kernel you may have to install the arm64 headers on your rpi4 to get arm64 kernel headers installed.

Use this script to do that:
https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51

caydenm, 2020-10-03:
I have the same challenge at OP. 

Openwrt module built using this [patch](https://github.com/1Jeff1/openwrt/commit/af426db0e78c3880a6bfe26fa97ae03caccce783.patch) by @1jeff1

I also made my own custom patch based on the above to try 5.7.0 branch in case recent fixes had been made which would rectify this issue. 

More info about the device and log lines is below. Happy to provide any more info to help debug this issue. 

Raspberry Pi 4b 2gb - Openwrt Dev snapshot from today
Wifi adapter: Simplecom NW602 
USB ID: 0bda:c811

With full driver logging on, I see: 

rtw_drv_log_level: 6 
RTW: module init start
RTW: rtw_inetaddr_notifier_register
usbcore: registered new interace driver rtl8812au 
RTW: module init ret=0
usb 1-1.4: new high-speed USB device number 7 using xhci_hcd 
usb 1-1.4: new USB device found, idVendor=obda, idProduct=c811, bcdDevice = 2.00

There is nothing after this in logread. 

Full device info from my desktop. 
Bus 003 Device 033: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xc811 
  bcdDevice            2.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11ac NIC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1

TheOneWhoseUsernameWasTaken, 2020-10-04:
Just as a quick update from my end: I couldn't get it to work, and the project deadline was approaching, so I spend another whopping 12 bucks and got myself a different wifi adapter. Now I'm using an Edimax EW-7711UAn V2 (usb id 7392:7710) which seems to use a Mediatek chipset (driver mt7601u). And it just works. Out of the box.

dash042, 2021-01-09:
I am actually having this issue with a ALFA AWUS036ACH with Arch Linux...
Anyone using Arch?