aircrack-ng/rtl8812au #723: Kali kernel updated and Awus1900 disappared

skiocink, 2020-09-18:
Hi to all,
today I upgraded Kali2020.2 to 5.8.0-kali1-amd64 and after a reboot I'm not able to use anymore my awus card.
Usually I use 3 different wireless cards and one of these (the awus1900) has as nickname: WIFI@REALTEK

root@kali2020:~# lsusb
...
Bus 001 Device 003: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
...

oot@kali2020:~# iwconfig
lo        no wireless extensions.

wlan0     IEEE 802.11  ESSID:"***************"  
          Mode:Managed  Frequency:2.457 GHz  Access Point: **:**:**:**:**:**   
          Bit Rate=90 Mb/s   Tx-Power=20 dBm   
          Retry short  long limit:2   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=34/70  Signal level=-76 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:421  Invalid misc:490   Missed beacon:0

wlan1     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

skiocink, 2020-09-18:
An upgarde...
Removing and installing again the 5.6.4.2 driver for kernel 5.7.0-kali3-amd64 the card works as usual.
Trying the same thing with kernel 5.8.0 produces this:

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.8.0-kali1-amd64 KSRC=/lib/modules/5.8.0-kali1-amd64/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.0-kali1-amd64 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.8.0-kali1-amd64 KSRC=/lib/modules/5.8.0-kali1-amd64/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.0-kali1-amd64 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.
Finished running dkms install steps.

kimocoder, 2020-10-09:
After kernel/modules upgrade, build driver again, with DKMS installation, this is not needed.

closed 2020-10-09 by kimocoder

Delson704557, 2020-11-17:
> After kernel/modules upgrade, build driver again, with DKMS installation, this is not needed.

Can you please elaborate the Solution??