aircrack-ng/rtl8812au #724: airodump-ng  802.11AC "lenght 0 "  Alpha AWUS036ACH Kali linux 2020.3

slt90, 2020-09-19:
Hi I have an alpha AWUS036ACH
[image]
 wifi card and I use Kali linux 2020.3 after installing the driver https://github.com/aircrack-ng/rtl8812au
i can detect both 2.4 and 5ghz networks but i have problems with 802.11ac frequencies.
If I use airodump-ng --band to wlan1 I don't detect the 802.11AC frequencies to find them I have to use
airodump-ng wlan1 -C 5100-5900, in this way I detect the AC networks (mac address and CH) but I can't see their ESSID. The network does not have a hidden ssd but airodump-ng cannot view it.
How can I solve the problem?
my test antennas are Ubiquiti rocket AC.
Thanks.

Mister-X-, 2020-09-19:
Could you post a PCAP with just of the beacons from that network?

slt90, 2020-09-19:
> Could you post a PCAP with just of the beacons from that network?

Hi Mister-X, sorry for the duplicate posts,I find this problem with a lot of wireless networks (I assume only 802.11ac networks) 

Here is the pcap file with a client connected to the network.
Thanks.

[testnetwork.pcap-01.zip](https://github.com/aircrack-ng/rtl8812au/files/5250823/testnetwork.pcap-01.zip)

Mister-X-, 2020-09-19:
The SSID isn't advertised. There is no SSID IE in the beacon at all.

slt90, 2020-09-19:
> The SSID isn't advertised. There is no SSID IE in the beacon at all.

> The SSID isn't advertised. There is no SSID IE in the beacon at all.

it happens to me only with 802.11 ac networks the name of this network and "test" as you can see from the photo it does not even find essid other networks

Mister-X-, 2020-09-20:
There is no "test" SSID on the picture. The -1 (channel) have been explained multiple times in the forum, and these are because a station sent a frame to an access point that is too far away from us (like a data frame, probe, whatever), and airodump-ng found a BSSID in the frame that isn't in its list, and shows it.

Upload a SINGLE beacon for the other networks so I can tell what is wrong. So far, this isn't a bug in airodump-ng. It could very well be a custom implementation in ubiquiti stuff.

slt90, 2020-09-20:
attached some beacon files, however I detect the problem only with AC networks.

[Network.zip](https://github.com/aircrack-ng/rtl8812au/files/5251376/Network.zip)

Mister-X-, 2020-09-20:
- ubnt-ultra SSID which has a tp-link MAC is a 802.11n network
- Ubiquiti MAC ending with 6c:93:23  5a:d8:7e are 802.11ac, no SSID IE in beacon (same as the one your posted initially
- Ubiquiti MAC ending with de:0e:eb is 802.11g, with UBNT ssid
- Ubiquiti MAC ending with 76:d4:49 is 802.11n, with hidden SSID IE

All the 802.11ac AP are ubiquiti. And as a matter of fact, all the beacons in these PCAP come from ubiquiti APs

It is now even more likely a custom implementation from ubiquiti, and not a bug. Unless you can show me it happens with a lot of other 802.11ac networks (that aren't ubiquiti), case closed.

closed 2020-09-20 by aircrack-ng

slt90, 2020-09-24:
can you add this ubiquiti customization to aircrack suite?
Thanks.

aircrack-ng, 2020-10-04:
If you can provide the information requested in aircrack-ng/aircrack-ng#2187