aircrack-ng/rtl8812au #725: No full monitor mode with Alpfa AWUS036ACH (rtl8812au)?

peterimnetz, 2020-09-21:
I compiled 5.6.4.2 and installed without any problems.

But in Wireshark I do not see some messages (e.g. EAPoL, DHCP) so I found that the checkbox in Wireshark for monitor mode can't be set (if I click I see the checkmark shortly but it disappears after 1s). This works fine for my Ralink RT5370, so I am woundering if this might be related to the rtl8812au driver for AWUS036ACH:

[image]

Both interfaces are in monitor mode:
[image]

I tried with pure tcpdump and got same results, looks like rtl8812au isn't in monitor mode? I tried with Kali as well - same behavior. Very strange, probably my mistake, but I am running out ouf ideas.

What is interesting in this context is the modprobe information, as for rtl8812au it just shows question marks.

constantinedev, 2020-09-26:
I have to same problem with 8812au AC usb devices problem
should airmon-ng can't take this deive's for now? 2020-09-26

peterimnetz, 2020-09-26:
I have those problems independant on airmon-ng, trying out a lot I found these "????????" in airmon-ng driver information, so my assumption was that this behavior may relate to the driver. Do you use a VM as well?

coffee44, 2020-10-26:
I've also been trying to fix the problem nearly for a week, but nothing has changed so far. Wireshark monitor mode checkbox cannot be checked.
In my case, Driver name is "88XXau", not like perterimnetz "????????".

peterimnetz, 2020-10-27:
I tried monitoring in same VM, but with RaLink RT5370 and RT5572 - these work perfectly, airlogs are complete. So I am quite sure that driver and/or RTL chipset have an issue.

vitorinux, 2020-11-02:
> I've also been trying to fix the problem nearly for a week, but nothing has changed so far. Wireshark monitor mode checkbox cannot be checked.
> In my case, Driver name is "88XXau", not like perterimnetz "????????".
> Do anyone know whether "wlanXmon" has to be created by the command "$airmon-ng start wlanX"?
> I could not create one.

I'm not sure but some module options on  "88XXau - Realtek Wireless Lan Driver"  set the parameter of renamed interfaces.

[image]

I'm using this wireless driver for a long, and the best option to put the interface in monitor mode is as mentioned in README.md

`
For setting monitor mode                                
  1. Fix problematic interference in monitor mode.
  ```                                                 
  $ airmon-ng check kill
  ```
  You may also uncheck the box "Automatically connect to this network when it is avaiable" in nm-connection-editor. This only works if you have a saved wifi connection.

  2. Set interface down
  ```
  $ sudo ip link set wlan0 down
  ```
  3. Set monitor mode
  ```
  $ sudo iw dev wlan0 set type monitor
  ```
  4. Set interface up
  ```
  $ sudo ip link set wlan0 up
```

`
AIRMON-NG is already coded to identify 88XXau drivers so the steps to put in monitor mode are different from others.

May be wireshark try to put interface on monitor mode in the old style, if this happens, a monX is created but nothing works on it.

Here a list of options of module parameters (I've used rtl8812au.v5.6.4.2 on a Raspbian 10 Buster, kernel 5.4.72-v7l+ armv7l)

###  88XXau - Realtek Wireless Lan Driver	

>   rtw_wireless_mode = 95
>      (int)
>   rtw_monitor_overwrite_seqnum = 0
>     Overwrite the sequence number of injected frames (int)
>   rtw_monitor_retransmit = 0
>     Retransmit injected frames (int)
>   rtw_monitor_disable_1m = 0
>     Disable default 1Mbps rate for monitor injected frames (int)
>   rtw_ips_mode = 0
>     The default IPS mode (int)
>   rtw_lps_level = 0
>     The default LPS level (int)
>   rtw_lps_chk_by_tp = 0
>      (int)
>   rtw_max_bss_cnt = 0
>      (int)
>   rtw_usb_rxagg_mode = 2
>      (int)
>   rtw_dynamic_agg_enable = 1
>      (int)
>   rtw_tx_bw_mode = 33
>     The max tx bw for 2.4G and 5G. format is the same as rtw_bw_mode (uint)
>   rtw_rx_ampdu_sz_limit_1ss = 
>     RX AMPDU size limit for 1SS link of each BW, 0xFF: no limitation (array of uint)
>   rtw_rx_ampdu_sz_limit_2ss = 
>     RX AMPDU size limit for 2SS link of each BW, 0xFF: no limitation (array of uint)
>   rtw_rx_ampdu_sz_limit_3ss = 
>     RX AMPDU size limit for 3SS link of each BW, 0xFF: no limitation (array of uint)
>   rtw_rx_ampdu_sz_limit_4ss = 
[... 152 lines cut ...]
>     5G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
>   rtw_target_tx_pwr_5g_b = 
>     5G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
>   rtw_target_tx_pwr_5g_c = 
>     5G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
>   rtw_target_tx_pwr_5g_d = 
>     5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
>   rtw_tsf_update_pause_factor = 200
>     num of bcn intervals to stay TSF update pause status (int)
>   rtw_tsf_update_restore_factor = 5
>     num of bcn intervals to stay TSF update restore status (int)
>   rtw_phy_file_path = /lib/firmware/
>     The path of phy parameter (charp)
>   rtw_load_phy_file = 68
>     PHY File Bit Map (int)
>   rtw_decrypt_phy_file = 0
>     Enable Decrypt PHY File (int)
>   rtw_en_napi = 1
>      (int)
>   rtw_en_gro = 1
>      (int)
>   rtw_iqk_fw_offload = 0
>      (int)
>   rtw_ch_switch_offload = 0
>      (int)

Spectre5, 2020-11-02:
I have the same issue as the OP, the check box for monitor mode in wireshard will not stay checked.  I've already set the adapter to monitor mode using the command provided in the README:

    $ sudo airmon-ng check kill
    $ sudo ip link set wlx00c0caac34bd  down
    $ sudo iw dev wlx00c0caac34bd set type monitor
    $ sudo ip link set wlx00c0caac34bd up

As an aside, I'm not sure why but the driver doesn't name the card wlan0, but rather wlx00c0caac34bd as shown in my commands above.  These commands appear to work fine as when I check with `iwconfig` it shows the interface being in monitor mode, but still wireshark will not keep the "monitor" check box checked.

I'm using wireshark 3.2.3 on native Ubuntu 20.04 and aircrack-ng tools version 1.6.  I've found that if I try using `airmon-ng start wlx00c0caac34bd` that the command freezes never returns, even if it is done after the using check kill.

n1tes, 2021-01-06:
have anyone found solution this?

Spectre5, 2021-01-06:
Basically, if you just put the wifi card into monitor mode outside of wireshark, then it still works --- even though wireshark will show monitor mode being unchecked.  Here is what I do, assuming that `wlx00c0caac34bd` is my Alfa card:
```
sudo airmon-ng check kill                      # may or may not be needed
sudo ip link set wlx00c0caac34bd down          # turn it off
sudo iw dev wlx00c0caac34bd set type monitor   # set card to monitor mode
sudo ip link set wlx00c0caac34bd up            # turn it back on
sudo iw dev wlx00c0caac34bd set channel 6      # set the channel here, not in wireshark
```

Then just start wireshark and select this card as the interface.  Don't try changing the channel nor selecting "monitor" mode within wireshark --- it is already in monitor mode even though wireshark doesn't realize it.  The channel part is important too, as wireshark won't change the channel for my card, so use the command above.

Occasionally you may have issues with the `up` command above.  If so, this may help:
```
# https://askubuntu.com/questions/62166/siocsifflags-operation-not-possible-due-to-rf-kill
# https://bbs.archlinux.org/viewtopic.php?pid=1324810#p1324810
sudo rfkill list
sudo rfkill unblock wifi; sudo rfkill unblock all
sudo rfkill list
sudo lshw -class network
```

When you are done, you can restart network-manager (or whatever else you need) which was killed with the `check kill` command above.  Using `systemctl` or `service` you would one of:
```
systemctl restart network-manager
sudo service NetworkManager restart
```

n1tes, 2021-01-06:
Hi @Spectre5 , thank you for the prompt response. I followed exactly what you said. No error but unfortunately nothing's changed. It is still in monitor mode but not capturing everything.

Spectre5, 2021-01-06:
Did you set the channel too?  I thought at one point that I wasn't getting everything but after setting the channel to my routers channel, I found everything.  The channel changing it wireshark will not work for me, the same as how it fails to check the monitor box.  If you tried this and it still doesn't work, then I'm out of ideas!  I do wish the card had better support

n1tes, 2021-01-06:
I did change the channel from command line (did not try in wireshark). Appreciate your help. I will update if I fix it. Below is my current driver though, just FYI
[image]