aircrack-ng/rtl8812au #726: did i make an error in make && make install

MiChaelinzo, 2020-09-22:
~/rtl8812au$ sudo make && make install
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.7.0-kali3-amd64/build M=/home/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.7.0-kali3-amd64'
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.7.0-kali3-amd64'
---------------------------------------------------------------------------
Visit https://github.com/aircrack-ng/rtl8812au for support/reporting issues
or check for newer versions (branches) of these drivers.                   
---------------------------------------------------------------------------
install -p -m 644 88XXau.ko  /lib/modules/5.7.0-kali3-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 5.7.0-kali3-amd64

**I first had an error having permission denied even with sudo command with it planning to install it in root but I just changed permission of the folder to 777 to make it work but I'm still wondering if my make && make install is successfull?**

MiChaelinzo, 2020-09-22:
**I first had an error having permission denied even with sudo command with it planning to install it in root but I just changed permission of the target directory to 777 to make it work but I'm still wondering if my make && make install is successful? I just want my TP-Link AC1300 MU-MIMO Archer T4U to work in my kali linux**

justinvforvendetta, 2020-09-22:
looks like it went fine.

MiChaelinzo, 2020-09-22:
**It does but I'm currently facing issues with permission ** 
[https://github.com/aircrack-ng/rtl8812au/issues/727](url)

MiChaelinzo, 2020-09-22:
```
sudo airmon-ng 

PHY     Interface       Driver          Chipset

phy0    wlan0           88XXau          TP-Link Archer T4U ver.3
```

MiChaelinzo, 2020-09-22:
> why did you make/install as root? why not with your normal user? you could have just chmod the entire git repo once you cloned it if u needed to

I just followed instruction here https://github.com/aircrack-ng/rtl8812au
and I'm not in root I'm in user while make/install

isneverdead, 2020-10-11:
mee too

`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.19.0-kali5-amd64/build M=/root/Music/download 2/kernel/drivers/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-kali5-amd64'
arch/x86/Makefile:150: CONFIG_X86_X32 enabled but no binutils support
make[3]: *** No rule to make target '2/kernel/drivers/rtl8812au'.  Stop.
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-kali5-amd64'
make: *** [Makefile:2246: modules] Error 2
`

hackingking466, 2020-11-03:
i have the same problem dude
i have the alfa AWUS1900 and i cant get it to work

hackingking466, 2020-11-03:
did it worked for u?

MiChaelinzo, 2020-11-12:
> did it worked for u?

yes it does

closed 2020-11-12 by MiChaelinzo