aircrack-ng/rtl8812au #729: Kernel 5.8 build errors on DKMS, works with MAKE

Muddleon, 2020-09-26:
I downloaded the current code with the kernel 5.8 fix.
**dkms-install** throws errors:

> DKMS make.log for rtl8812au-5.6.4.2 for kernel 5.8.8-100.fc31.x86_64 (x86_64)
>       2 Sat 26 Sep 2020 03:29:14 PM CDT
>       3 make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.8-100.fc31.x86_64/build M=/var/lib/dkms/rtl8812au/5.6.4.2/build  modules
>       4 make[1]: Entering directory '/usr/src/kernels/5.8.8-100.fc31.x86_64'
>       5   CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_cmd.o
>       6   CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.o
>       7   CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_debug.o
>       8   CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_io.o
>       9   CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ioctl_query.o
>      10   CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ioctl_set.o
>      11   CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ieee80211.o
>      12   CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme.o
>      13 In file included from /var/lib/dkms/rtl8812au/5.6.4.2/build/include/drv_types.h:70,
>      14                  from /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_io.c:47:
>      15 /var/lib/dkms/rtl8812au/5.6.4.2/build/include/rtw_security.h:257:8: error: redefinition of ‘struct sha256_state’
>      16   257 | struct sha256_state {
>      17       |        ^~~~~~~~~~~~
>      18 In file included from /var/lib/dkms/rtl8812au/5.6.4.2/build/include/drv_types.h:70,
>      19                  from /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_debug.c:17:
>      20 /var/lib/dkms/rtl8812au/5.6.4.2/build/include/rtw_security.h:257:8: error: redefinition of ‘struct sha256_state’
>      21   257 | struct sha256_state {
>      22       |        ^~~~~~~~~~~~

Fortunately plain _make_ and _make install_ appears to work.

Lauriy, 2020-10-05:
Successfully did DKMS install on 5.8.13 with 5.7.0 branch.

kimocoder, 2020-10-09:
git pull and rebuild.

closed 2020-10-09 by kimocoder

iclanzan, 2020-10-18:
The v5.7.0 branch successfully compiled for me, while the latest v5.6.4.2 is still failing. Kernel version is 5.8.8.
What is the difference between the `v5.7.0`  and `v5.6.4.2`?