aircrack-ng/rtl8812au #736: Warnings in kernel message when hopping channels with Kismet+Kali

ttimasdf, 2020-09-30:
The logs start to pop up continuously in dmesg after kismet is launched. No matter I start it with one fixed channel or default setting (hopping through every 2.4/5G channels).

According to the log message, the warning was throwed at here.

https://github.com/aircrack-ng/rtl8812au/blob/07c704c0a7131208a909c3fc36e7daa122b98b16/core/rtw_rf.c#L214-L216

which is called from here.

https://github.com/aircrack-ng/rtl8812au/blob/07c704c0a7131208a909c3fc36e7daa122b98b16/hal/hal_intf.c#L813-L814

environment:

```
➜  rtl8812au git:(v5.6.4.2) ✗ uname -a              
Linux kali 5.8.0-kali2-amd64 #1 SMP Debian 5.8.10-1kali1 (2020-09-22) x86_64 GNU/Linux
➜  rtl8812au git:(v5.6.4.2) ✗ lsb_release -a
No LSB modules are available.
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2020.3
Codename:       kali-rolling
```

related dmesg:

```
[ 6113.003126] ------------[ cut here ]------------
[ 6113.003197] WARNING: CPU: 3 PID: 56785 at /home/user/Tools/rtl8812au/core/rtw_rf.c:215 rtw_get_scch_by_cch_offset+0x59/0x80 [88XXau]
[ 6113.003208] Modules linked in: 88XXau(OE) nfnetlink_queue(E) nfnetlink_log(E) nfnetlink(E) bluetooth(E) jitterentropy_rng(E) drbg(E) ansi_cprng(E) ecdh_generic(E) ecc(E) cfg80211(E) rfkill(E) intel_rapl_msr(E) vsock_loopback(E) vmw_vsock_virtio_transport_common(E) intel_rapl_common(E) vmw_vsock_vmci_transport(E) vsock(E) rapl(E) vmw_balloon(E) binfmt_misc(E) nls_ascii(E) nls_cp437(E) vfat(E) efi_pstore(E) fat(E) snd_ens1371(E) joydev(E) snd_ac97_codec(E) serio_raw(E) pcspkr(E) efivars(E) ac97_bus(E) gameport(E) snd_rawmidi(E) snd_seq_device(E) snd_pcm(E) snd_timer(E) snd(E) soundcore(E) vmw_vmci(E) sg(E) evdev(E) ac(E) fuse(E) sunrpc(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) ext4(E) crc16(E) mbcache(E) jbd2(E) crc32c_generic(E) sd_mod(E) t10_pi(E) crc_t10dif(E) crct10dif_generic(E) hid_generic(E) usbhid(E) hid(E) sr_mod(E) cdrom(E) ata_generic(E) crct10dif_pclmul(E) crct10dif_common(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) vmwgfx(E) aesni_intel(E) ttm(E) libaes(E)
[ 6113.003256]  crypto_simd(E) cryptd(E) glue_helper(E) xhci_pci(E) drm_kms_helper(E) ata_piix(E) mptspi(E) ehci_pci(E) uhci_hcd(E) cec(E) mptscsih(E) xhci_hcd(E) psmouse(E) mptbase(E) ehci_hcd(E) scsi_transport_spi(E) libata(E) drm(E) usbcore(E) scsi_mod(E) e1000(E) usb_common(E) i2c_piix4(E) button(E) [last unloaded: 88XXau]                                                                                                                                              
[ 6113.003281] CPU: 3 PID: 56785 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.8.0-kali2-amd64 #1 Debian 5.8.10-1kali1
[ 6113.003324] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.16722896.B64.2008100651 08/10/2020
[ 6113.003379] RIP: 0010:rtw_get_scch_by_cch_offset+0x59/0x80 [88XXau]
[ 6113.003393] Code: 8d 47 d6 3c 81 77 06 40 80 fe 02 74 2f 8d 47 da 3c 89 77 11 84 c9 74 0d 8d 47 02 83 ef 02 80 fa 02 0f 45 c7 c3 0f 0b 31 c0 c3 <0f> 0b 31 c0 c3 8d 47 08 83 ef 08 80 fa 02 0f 45 c7 c3 8d 47 04 83
[ 6113.003404] RSP: 0018:ffffb9344279fe08 EFLAGS: 00010246
[ 6113.003415] RAX: 0000000000000036 RBX: 000000000000003a RCX: 0000000000000000
[ 6113.003426] RDX: 0000000000000000 RSI: 0000000000000001 RDI: 0000000000000036
[ 6113.003436] RBP: ffffb93440d01000 R08: 0000000000000001 R09: 0000000000000000
[ 6113.003447] R10: 0000058f4bef5279 R11: 0000000000000000 R12: 0000000000000002
[ 6113.003457] R13: 000000000000003a R14: 0000000000000036 R15: ffffb934404c3000
[ 6113.003469] FS:  0000000000000000(0000) GS:ffff90161c6c0000(0000) knlGS:0000000000000000
[ 6113.003480] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 6113.003490] CR2: 000000000213400e CR3: 000000004f2e4006 CR4: 00000000003606e0
[ 6113.003503] Call Trace:
[ 6113.003572]  rtw_hal_set_chnl_bw+0xf0/0x130 [88XXau]
[ 6113.003625]  set_channel_bwmode+0xc6/0x110 [88XXau]
[ 6113.003673]  rtw_set_chbw_hdl+0x4b/0x60 [88XXau]
[ 6113.003710]  rtw_cmd_thread+0x200/0x530 [88XXau]
[ 6113.003758]  ? rtw_set_external_auth_status+0xc0/0xc0 [88XXau]
[ 6113.003795]  ? rtw_stop_cmd_thread+0x40/0x40 [88XXau]
[ 6113.003810]  kthread+0x11b/0x140
[ 6113.003823]  ? __kthread_bind_mask+0x60/0x60
[ 6113.003836]  ret_from_fork+0x22/0x30
[ 6113.003849] ---[ end trace 72e15264770f712c ]---
```

cain986, 2020-11-20:
I'm having this same issue. It looks like it's only a warning, so I'm not sure if I should care about this or not.

ttimasdf, 2021-01-04:
Kismet works after some rebooting so I'm also unsure about that.. whether it's a warning or a panic.

fpusersuggest, 2023-06-13:
hello, here I've the same problem with rpi3b+, raspbian bullseye, kernel 6.1.21, and kismet.
After some minutes, around 5-10 minutes. Kismet stop working, I mean it stop to receive data from teh device. I have to restart it.
I used branch 5.6.4.2 and now I'll try 4.3.21.