aircrack-ng/rtl8812au #738: RTL8812AE - Slow speeds - 2.4/5G

ShayBox, 2020-10-03:
My laptop (Lenovo Ideapad 110-15ISK) has a built-in RTL8812AE card, it has issues picking up any networks after initial boot, and once its started working, upload is around 25mbps and download is around 70mbps, I can't figure out how to fix this, i've tried the kernel driver, v5.6.x.4, and v5.7.0, 2.4 and 5g, two routers, all the same.

This is my iperf3 output from my laptop to my router
```
 ~  iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.103 port 43460 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  3.67 MBytes  30.8 Mbits/sec    0    164 KBytes       
[  5]   1.00-2.00   sec  3.54 MBytes  29.7 Mbits/sec    0    232 KBytes       
[  5]   2.00-3.00   sec  3.36 MBytes  28.1 Mbits/sec    0    283 KBytes       
[  5]   3.00-4.00   sec  2.98 MBytes  25.0 Mbits/sec    0    300 KBytes       
[  5]   4.00-5.00   sec  3.11 MBytes  26.1 Mbits/sec    0    300 KBytes       
[  5]   5.00-6.00   sec  3.11 MBytes  26.1 Mbits/sec    0    300 KBytes       
[  5]   6.00-7.00   sec  3.11 MBytes  26.1 Mbits/sec    0    300 KBytes       
[  5]   7.00-8.00   sec  3.11 MBytes  26.1 Mbits/sec    0    300 KBytes       
[  5]   8.00-9.00   sec  3.29 MBytes  27.6 Mbits/sec    0    317 KBytes       
[  5]   9.00-10.00  sec  2.67 MBytes  22.4 Mbits/sec    0    317 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  31.9 MBytes  26.8 Mbits/sec    0             sender
[  5]   0.00-10.03  sec  30.8 MBytes  25.8 Mbits/sec                  receiver

iperf Done.
```

stribuda, 2020-12-24:
I have the same problem but even worse speeds (around 6Mbits..)
No one has an solution?

zavndw, 2021-01-17:
RTL8812AE
5.4.89-1-lts
speed upload 5 Mbits/sec

ShayBox, 2021-01-18:
I replaced my card, I don't know what the problem was but ir was unusable

zavndw, 2021-01-18:
> I replaced my card, I don't know what the problem was but it was unusable
> 
> Cheap on ebay but takes awhile for delivery

Which one did you replace?

ShayBox, 2021-01-19:
Intel Wireless-AC 8265 $15 total, works flawlessly, no having to toggle airplane mode to get wifi working or slow speeds, anything Intel.

zavndw, 2021-01-19:
> Intel Wireless-AC 8265 $15 total, works flawlessly, no having to toggle airplane mode to get wifi working or slow speeds, anything Intel.

thx

imyxh, 2022-01-21:
I'm confused; is the rtl8812ae card even able to run using the driver for the rtl8812a**u**? I have an rtl8812ae and it normally uses the rtl88**21**ae driver from the kernel. When I insert the 88XXau module, and remove the rtl8821ae module, my network card then shows up in lshw as unclaimed.

My understanding was that this repository is only for the USB dongle rtl8812a**u**, and not for the PCIe card rtl8812a**e**?