aircrack-ng/rtl8812au #740: Crashes for 3x3:3 AC1300

haraldrudell, 2020-10-03:
Works as AP but not as STA
Other slower 2x2 STA work

Have an access point 5 GHz 80 MHz 3x3:3 running `iperf3 --server`
https://github.com/haraldrudell/AC1300-USB/blob/master/README.md

Have a usb3 computer with device up in 5 Mb/s mode:

```
apt install --yes build-essential git dkms
cd ~
git clone --depth 1 https://github.com/aircrack-ng/rtl8812au.git
cd ~/rtl8812au
./dkms-install.sh
modprobe 88XXau rtw_switch_usb_mode=1
cat /sys/module/88XXau/version
v5.6.4.2_35491.20191025
cat /sys/module/88XXau/parameters/rtw_switch_usb_mode 
1
lsusb -t | grep rtl88XXau
        |__ Port 3: Dev 4, If 0, Class=Vendor Specific Class, Driver=rtl88XXau, 5000M
ip l s wlx08beac12be7a up
iw dev wlx08beac12be7a connect entebra
ip a s wlx08beac12be7a
dhclient wlx08beac12be7a
iperf3 --client 10.36.0.1
Connecting to host 10.36.0.1, port 5201
[  5] local 10.36.0.56 port 53610 connected to 10.36.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  83.9 KBytes   686 Kbits/sec    2   2.21 KBytes       
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    1   2.21 KBytes       
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    0   2.21 KBytes       
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   2.21 KBytes       
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    0   2.21 KBytes       
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    0   2.21 KBytes       
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    1   2.21 KBytes       
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    0   2.21 KBytes       
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   2.21 KBytes       
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    0   2.21 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  83.9 KBytes  68.7 Kbits/sec    5             sender
[  5]   0.00-10.00  sec  0.00 Bytes  0.00 bits/sec                  receiver
uname --all
Linux c89x1 5.3.0-40-generic #32-Ubuntu SMP Fri Jan 31 20:24:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
lsb_release --all
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 19.10
Release:        19.10
Codename:       eoan
```

`less /var/log/syslog`
```
Oct  3 15:22:19 c89x1 kernel: [ 1525.379859] ------------[ cut here ]------------
Oct  3 15:22:19 c89x1 kernel: [ 1525.379864] URB ffff99c648c11900 submitted while active
Oct  3 15:22:19 c89x1 kernel: [ 1525.379888] WARNING: CPU: 3 PID: 9841 at drivers/usb/core/urb.c:363 usb_submit_urb+0x4f1/0x5e0
Oct  3 15:22:19 c89x1 kernel: [ 1525.379889] Modules linked in: 88XXau(OE) rfcomm bnep binfmt_misc zfs(PO) zunicode(PO) zavl(PO) icp(PO)
 zlua(PO) nls_iso8859_1 zcommon(PO) znvpair(PO) spl(O) dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua snd_hda_codec_cirrus snd_hda_c
odec_generic ledtrig_audio snd_hda_codec_hdmi intel_rapl_msr snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_pcm mei_hdcp joydev 
snd_seq_midi snd_seq_midi_event snd_rawmidi intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel snd_seq btusb kvm
 applesmc btrtl input_polldev btbcm btintel irqbypass intel_cstate intel_rapl_perf bluetooth snd_seq_device ecdh_generic input_leds ecc 
intel_pch_thermal snd_timer brcmfmac brcmutil cdc_ether usbnet r8152 bcm5974 mii snd cfg80211 mei_me bdc_pci spi_pxa2xx_pci mei dw_dmac_
pci soundcore applespi sbs acpi_als kfifo_buf spi_pxa2xx_platform dw_dmac industrialio dw_dmac_core sbshc apple_bl mac_hid sch_fq_codel 
parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs zstd_compress dm_crypt
Oct  3 15:22:19 c89x1 kernel: [ 1525.379958]  raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq lib
crc32c raid0 multipath linear dm_mirror dm_region_hash dm_log raid1 hid_apple hid_generic usbhid hid uas usb_storage crct10dif_pclmul cr
c32_pclmul ghash_clmulni_intel i915 aesni_intel i2c_algo_bit drm_kms_helper syscopyarea sysfillrect aes_x86_64 crypto_simd cryptd sysimg
blt fb_sys_fops glue_helper i2c_i801 ahci libahci drm lpc_ich thunderbolt video [last unloaded: 88XXau]
[... 36 lines cut ...]
Oct  3 15:22:19 c89x1 kernel: [ 1525.380456]  __tcp_transmit_skb+0x549/0xb00
Oct  3 15:22:19 c89x1 kernel: [ 1525.380460]  tcp_write_xmit+0x3bc/0xb90
Oct  3 15:22:19 c89x1 kernel: [ 1525.380465]  tcp_push_one+0x3c/0x40
Oct  3 15:22:19 c89x1 kernel: [ 1525.380468]  tcp_sendmsg_locked+0x75c/0xd60
Oct  3 15:22:19 c89x1 kernel: [ 1525.380473]  tcp_sendmsg+0x2d/0x50
Oct  3 15:22:19 c89x1 kernel: [ 1525.380478]  inet_sendmsg+0x43/0x70
Oct  3 15:22:19 c89x1 kernel: [ 1525.380482]  sock_sendmsg+0x5e/0x70
Oct  3 15:22:19 c89x1 kernel: [ 1525.380486]  sock_write_iter+0x8c/0xf0
Oct  3 15:22:19 c89x1 kernel: [ 1525.380491]  new_sync_write+0x125/0x1c0
Oct  3 15:22:19 c89x1 kernel: [ 1525.380495]  __vfs_write+0x29/0x40
Oct  3 15:22:19 c89x1 kernel: [ 1525.380498]  vfs_write+0xb9/0x1a0
Oct  3 15:22:19 c89x1 kernel: [ 1525.380502]  ksys_write+0x67/0xe0
Oct  3 15:22:19 c89x1 kernel: [ 1525.380506]  __x64_sys_write+0x1a/0x20
Oct  3 15:22:19 c89x1 kernel: [ 1525.380511]  do_syscall_64+0x5a/0x130
Oct  3 15:22:19 c89x1 kernel: [ 1525.380516]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Oct  3 15:22:19 c89x1 kernel: [ 1525.380519] RIP: 0033:0x7f546f9af317
Oct  3 15:22:19 c89x1 kernel: [ 1525.380522] Code: 64 89 02 48 c7 c0 ff ff ff ff eb bb 0f 1f 80 00 00 00 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 48 89 54 24 18 48 89 74 24
Oct  3 15:22:19 c89x1 kernel: [ 1525.380524] RSP: 002b:00007ffdb03ce8e8 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
Oct  3 15:22:19 c89x1 kernel: [ 1525.380527] RAX: ffffffffffffffda RBX: 0000000000020000 RCX: 00007f546f9af317
Oct  3 15:22:19 c89x1 kernel: [ 1525.380529] RDX: 0000000000020000 RSI: 00007f546f576000 RDI: 0000000000000005
Oct  3 15:22:19 c89x1 kernel: [ 1525.380530] RBP: 00007f546f576000 R08: 0000000000000000 R09: 00007ffdb03ce978
Oct  3 15:22:19 c89x1 kernel: [ 1525.380532] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000005
Oct  3 15:22:19 c89x1 kernel: [ 1525.380533] R13: 0000000000020000 R14: 00007ffdb03ce940 R15: 000055cfc4f902a0
Oct  3 15:22:19 c89x1 kernel: [ 1525.380537] ---[ end trace 49f08ae82d0f0cdf ]---
```

haraldrudell, 2020-10-03:
macOS client bcm43602 reports channel 878 Mb/s mcs 7: 3x3 80 MHz
iperf3 throughput 451 Mb/s
and AP does not crash…

haraldrudell, 2020-10-04:
echo "options 88XXau rtw_switch_usb_mode=1 rtw_led_ctrl=0" >/etc/modprobe.d/8812au.conf
…
cat /proc/net/rtl88XXau/wlx08beac12be7a/led_ctrl 
0
cat /sys/module/88XXau/parameters/rtw_switch_usb_mode
1
no change

haraldrudell, 2020-10-04:
the STA works agains 802.11g 54 Mb/s: iperf3 18.5 Mb/s

memnoth, 2020-10-07:
Just a workaround patch for that case, but yet 8814 has a performance issue though.

https://github.com/net-rw/rtl8812au/issues/1

haraldrudell, 2020-10-10:
This driver, which claims to be 5.8.5 does work at 469 Mb/s:
https://github.com/coolshou/rtl8814au

Basically:
<pre>
# rmmod and make uninstall or ./dkms-remove.sh of whatever was there before
cd ~
git clone --depth 1 https://github.com/coolshou/rtl8814au.git coolshou
cd coolshou/driver
nano ./os_dep/linux/usb_intf.c
# 191 is: {USB_DEVICE(0x2001, 0x331a), .driver_info = RTL8814A}, /* D-Link - D-Link */
# add after line 191:
        {USB_DEVICE(0x7392, 0xA833), .driver_info = RTL8814A}, /* Edimax - AC1750 */
#
echo "options 8814au rtw_switch_usb_mode=1" >/etc/modprobe.d/8814au.conf
make
make install
modprobe 8814au
cat /sys/module/8814au/version
5.8.5-34085
cat /sys/module/8814au/parameters/rtw_switch_usb_mode
1
# check what devices driver claims:
modprobe -c | egrep '8814au$'
alias usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in* 8814au
alias usb:v2001p331Ad*dc*dsc*dp*ic*isc*ip*in* 8814au
alias usb:v7392pA833d*dc*dsc*dp*ic*isc*ip*in* 8814au
# plug in Wi-Fi device
lsusb -t | grep 8814au
        |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=rtl8814au, 5000M
ip l | grep wlx
8: wlx08beac12be7a: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
iw dev
               phy#4
                       Interface wlx08beac12be7a
                               ifindex 7
                               wdev 0x400000001
                               addr 08:be:ac:12:be:7a
                               type managed
                               txpower 20.00 dBm
               phy#0
                       Interface wlp3s0
                               ifindex 4
                               wdev 0x1
                               addr f4:5c:89:8a:29:2f
                               type managed
                               channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
# device is phy4
rfkill
ID TYPE      DEVICE    SOFT      HARD
 0 bluetooth hci0   blocked unblocked
 2 wlan      phy0   blocked unblocked
 5 wlan      phy4   blocked unblocked
# id is 5
rfkill unblock 5
#
# configure interface
ip l sh wlx08beac12be7a
# should be state UP mode DEFAULT
ip l s wlx08beac12be7a up
iw wlx08beac12be7a connect entebbe
iw wlx08beac12be7a link
Connected to 08:be:ac:12:be:74 (on wlx08beac12be7a)
        SSID: entebbe
        freq: 5180
        signal: -48 dBm
        tx bitrate: 1300.0 MBit/s
# 1300 is what we want to see
ip a s wlx08beac12be7a
dhclient wlx08beac12be7a
ip r del 0/0 via 10.36.0.1
iperf3 --client 10.36.0.1
472 Mbits/sec    0             sender
469 Mbits/sec                  receiver
😄😄😄
</pre>

aircrack-ng is broken somehow but did take code from this repo

Now, gentlemen, you have a working and a failing case, and we know it can be done!

haraldrudell, 2020-10-10:
> Just a workaround patch for that case, but yet 8814 has a performance issue though.
> 
> [net-rw#1](https://github.com/net-rw/rtl8812au/issues/1)

Applying that patch removes the crash but does not increase throughput. Bug still there

kimocoder, 2020-10-10:
@haraldrudell Yeah, finally a more clean driver for that chipset also, rtl8812au and rtl8821au is quite similar, rtl8814au code never was..

@evilphish Here's some news for us 🥇

kimocoder, 2020-10-10:
Ok, this is a driver developed by Edimax actually. Vedpr based.

Here's the link to ZIP below;
https://www.edimax.com/edimax/mw/cufiles/files/download/Driver_Utility/EW-7833UAC/EW-7833UAC_linux_5.8.5.1_kernel_2.6-5.1.zip

What I'll do, I'll compare it to the past release of 8814au and see what changed. Damn!

haraldrudell, 2020-10-10:
Realtek rtl8814au from 2010 is Wi-Fi 5 802.11ac 4x4:3 AC1900: 1.3 Gb/s on 5 GHz 80 MHz and 600 Mb/s on 2.4 GHz 40 MHz

That makes it the fastest usb chip at present until 4x4:4 or Wi-Fi 6 802.11ax is available. And it is supported or at least working on Linux/macOS/Windows. The maximum speed calls for a usb3 port at 5 Gb/s or better

At least for Linux, Access Point and Wi-Fi Direct is supported, too. That means we have a dual band eap-tls or wpa3 with continuous security updates

NETGEAR A7000 seems to have put the best antennas on it but there are many like: D-Link DWA-192, Edimax EW7833UAC, TP-Link Archer T9UH

kimocoder, 2020-10-11:
Tube-UAC2 with directional panel.. beats everything. Like a magic wand 👍 

[image]

haraldrudell, 2020-10-18:
Some additional differences:

Neither driver supports multiple ssids:
**iw list**
interface combinations are not supported

The coolshou driver does not have the radar detection channels 52…144
but has channels 144…177 disabled, 14 disabled

coolshou driver does not have:
HT 2.4/5 GHz:
RX Greenfield
RX STBC 3-streams
VHT 5 GHz:
> [RX-STBC-1]
ie. the coolshou driver does not:
— offer non-backwards compatible, higher performing HT channel: 2.4 GHz 802.11n performance
— receive multiple spatial streams

coolshou does not have command:
set_wiphy_netns

The conclusion is, as of October 17th, 2020:
use 88XXau for AP
use 8814au for STA

haraldrudell, 2020-10-18:
I am not sure if those software capability descriptions is what the hardware is actually doing

The coolshou driver could receive at 469 Mb/s on 5 GHz as verified by iperf3. That speed will only be attained when using all 3 spatial streams

haraldrudell, 2020-10-25:
It seems that SAE WPA3-Personal does not work as STA or AP
8814 does not have protected management frames, but does have SAE
**iw list**
…
MFP-capable # is missing
Device supports SAE with AUTHENTICATE command

Because the driver has no combinations,
multi-band AP or
multi-ssid authentication like guest network is unavailable

haraldrudell, 2020-10-26:
For reference, here is **iw list** output for the 88XXau aircrack-ng driver and rtl8814au. This data can be difficult to come by

— multiple ssids is not supported “interface combinations are not supported” so multi-band AP or guest network involves adding more adapters
— protected management frames is not supported 802.11w, so it will not work with certified WPA3 hardware that will reject this device: iw list “MFP-capable”
— SAE, ie. WPA3 authentication, is supported “Device supports SAE with AUTHENTICATE command”
— — SAE is as of now the only safe password-based Wi-Fi of any kind
— — SAE makes password attack unfeasible for passwords that are difficult to guess
— — SAE mutually authenticates STA and AP like eap-tls that must both have the secret
— — Wi-Fi security that is not eap-tls prior to 2018 is garbage as is much embedded software out of China

The best use of this hardware is 469 Mb/s 5 GHz 80 MHz eap-tls. We’ll all enjoy very safe amazing performance. No known vulnerabilities whatsoever since 0 BC

<pre>
Wiphy phy3                            
  max # scan SSIDs: 9                       
  max scan IEs length: 2304 bytes                 
  max # sched scan SSIDs: 0                       
  max # match sets: 0                       
  max # scan plans: 1                       
  max scan plan interval: -1                      
  max scan plan iterations: 0                     
  Retry short limit: 7                      
  Retry long limit: 4                       
  Coverage class: 0 (up to 0m)                    
  Supported Ciphers:                        
    * WEP40 (00-0f-ac:1)                    
    * WEP104 (00-0f-ac:5)                   
    * TKIP (00-0f-ac:2)                     
    * CCMP-128 (00-0f-ac:4)                 
    * CMAC (00-0f-ac:6)                     
  Available Antennas: TX 0 RX 0                   
  Supported interface modes:                      
     * IBSS                     
     * managed                        
     * AP                       
     * monitor                        
  Band 1:                       
    Capabilities: 0x1963                    
      RX LDPC                   
      HT20/HT40                       
      Static SM Power Save                  
      RX HT20 SGI                     
      RX HT40 SGI                     
      RX STBC 1-stream                
      Max AMSDU length: 7935 bytes                
      DSSS/CCK HT40                   
    Maximum RX AMPDU length 65535 bytes (exponent: 0x003)           
    Minimum RX AMPDU time spacing: 16 usec (0x07)             
    HT Max RX data rate: 450 Mbps                 
    HT TX/RX MCS rate indexes supported: 0-23                 
    Bitrates (non-HT):                      
      * 1.0 Mbps                      
      * 2.0 Mbps                      
      * 5.5 Mbps                      
      * 11.0 Mbps                     
      * 6.0 Mbps                      
      * 9.0 Mbps                   [90/1951]
      * 12.0 Mbps
      * 18.0 Mbps
      * 24.0 Mbps
      * 36.0 Mbps
      * 48.0 Mbps
      * 54.0 Mbps
    Frequencies:
      * 2412 MHz [1] (20.0 dBm)
      * 2417 MHz [2] (20.0 dBm)
      * 2422 MHz [3] (20.0 dBm)
      * 2427 MHz [4] (20.0 dBm)
      * 2432 MHz [5] (20.0 dBm)
      * 2437 MHz [6] (20.0 dBm)
[... 102 lines cut ...]
     * connect
     * disconnect
  Supported TX frame types:
     * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
     * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
     * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
     * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
     * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
     * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
  Supported RX frame types:
     * IBSS: 0xd0
     * managed: 0x40 0xb0 0xd0
     * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
     * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
     * P2P-client: 0x40 0xd0
     * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
  WoWLAN support:
     * wake up on anything (device continues operating normally)
  software interface modes (can always be added):
     * monitor
  interface combinations are not supported
  Device supports SAE with AUTHENTICATE command
  Device supports scan flush.
  Supported extended features:
</pre>