aircrack-ng/rtl8812au #743: `make dkms_install` does not work on a fresh Pop!_OS installation

viniciussbs, 2020-10-11:
Hi. I've just installed Pop!_OS and followed this instructions:

```
sudo apt update
sudo apt install dkms
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au
sudo ./dkms-install.sh
```
After I saw the recent commit that removes the script, I've tried the following:

```
make dkms_install
```
...and got this:

```
Makefile:2300: *** Chamada não terminada para a função 'shell': faltando ')'.  Pare.
```
It's a portuguese message telling that there's a missing `)` in the `shell` command, but that's not true. The command looks OK:
```
$(shell grep "#define DRIVERVERSION" include/rtw_version.h | awk '{print $$3}' | tr -d v\")
```
But it seems I don't have the `shell` command (it says _"command `shell` not found"_):
```
Comando 'shell' não encontrado, você quis dizer:

  comando 'spell' do deb spell (1.0-24build1)
  comando 'shelr' do deb shelr (0.16.3-2)
  comando 'jshell' do deb openjdk-11-jdk-headless (11.0.8+10-0ubuntu1~20.04)
  comando 'jshell' do deb openjdk-13-jdk-headless (13.0.3+3-1ubuntu2)
  comando 'jshell' do deb openjdk-14-jdk-headless (14.0.1+7-1ubuntu1)
  comando 'bshell' do deb avahi-ui-utils (0.7-4ubuntu7)

Experimente: sudo apt install <deb name>
```
Do I need to install any dependency?

I've got it working rolling back to the commit before this change, so thank you for the driver.

kimocoder, 2020-10-11:
make && make install

viniciussbs, 2020-10-11:
I thought I should use `make dkms_install` instead of `make` based on this comment: https://github.com/aircrack-ng/rtl8812au/commit/280137863b08956c442ce0d3e67669ede781ee1c#commitcomment-43140919 Closing the issue, then. Thanks.

closed 2020-10-11 by viniciussbs