aircrack-ng/rtl8812au #744: Makefile is garbage

S4RK0Z1, 2020-10-11:
Whoever did change the ./dkms-install.sh method to this Makefile garbage, ty. Install was a piece of cake with the ./dkms-install.sh method. Right now I wasted 1 1/2 hours to try to get this Makefile work with no avail. Thank you for fcking up a perfectly working/unbroken thing to an unworking garbage.

kimocoder, 2020-10-11:
So you used 1,5 hours to write "make dkms_install" ?
Congrats!

closed 2020-10-11 by kimocoder

isneverdead, 2020-10-11:
can you send me the code?

> Whoever did change the ./dkms-install.sh method to this Makefile garbage, ty. Install was a piece of cake with the ./dkms-install.sh method. Right now I wasted 1 1/2 hours to try to get this [...]

hi, can you send me the code please?