aircrack-ng/rtl8812au #745: Injection doesn't work Alfa AWUS1900 (rtl8814au)

nidhoggDrag, 2020-10-11:
awus 1900(rtl8814au) driver rtl8812au, installed strictly according to the instructions

Hello! correct the injection please! The injection doesn't work!

when I write the command:

root@kali:~# aireplay-ng -9 wlan0
17:32:30  Trying broadcast probe requests...
17:32:30  Injection is working!
17:32:32  Found 8 APs
17:32:32  Trying directed probe requests...
17:32:32  xx:xx:xx:xx:xx:xx - channel: 9 - 'xxxxx'
17:32:33  Ping (min/avg/max): 7.657ms/15.522ms/21.115ms Power: -70.13
17:32:33  30/30: 100%

shows that the injection works, BUT! when you try to disconnect the device, then:

root@kali:~# aireplay-ng --deauth 20 -a xx:xx:xx:xx:xx:xx -c xx:xx:xx:xx:xx:xx wlan0
17:38:47  Waiting for beacon frame (BSSID: xx:xx:xx:xx:xx:xx) on channel 9                                                 
17:38:47  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]                                     
17:38:48  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]                                     
17:38:48  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]                                     

packages 0-0 - [0 | 0 ACKs] !!! Injection doesn't work

kimocoder, 2020-10-11:
This chipset has a new driver available and will be separated from this repo, only 8811, 8812 and 8821 Will be supported. I'll let you know, shortly

nidhoggDrag, 2020-10-11:
Hello! many thanks! otherwise I have been trying to overcome this problem for a
month =)

  11.10.2020, 17:09, Christian Bremvåg <notifications@github.com>

  This chipset has a new driver available and will be separated from this repo,
  only 8811, 8812 and 8821 Will be supported. I'll let you know, shortly