aircrack-ng/rtl8812au #748: Packet injection not working with RTL8812AU

justincase82, 2020-10-12:
I'm running Kali 2020 and installed the latest drivers etc. 

After some testing aireplay-ng get's zero ACKS when attempting a deauth.

I dug out my old RTL8187B put it into monitor mode and it injects and de auths without a hitch. 

Am I missing something here or are these chipsets hit or miss? Should I just return it and get something else?

justincase82, 2020-10-12:
used DKM package no issues

closed 2020-10-12 by justincase82

Craftonix, 2020-11-07:
Hello,
what do you mean by "used DKM package"
i have same issue, found that using iwconfig instead of iw to set in monitor mode is better but not sure.
What do you use to set in monitor mode ?

justincase82, 2020-11-07:
If I download / compile / install the github package manually the driver
loads but injection did not work at all..

If I apt install realtek-rtl88xxau-dkms the everything works perfectly,
same driver versions too... not sure why.

Craftonix, 2020-11-09:
Thanks for the reply,
I finally done it this way :
Git clone 
sudo make dkms_install

make / make install which i used first made the injection fail.
next time i will install with with your method.