aircrack-ng/rtl8812au #749: Syntax error occurred in Makefile

mose-s, 2020-10-13:
`make dkms_install` gave me the following error:
> Makefile:2300: *** unterminated call to function 'shell': missing ')'.  Stop.

I think the # is interpreted as the start of a comment, so I escaped the character and got it fixed

mastni, 2020-10-13:
I can confirm this happens on my end as well, and @mose-s's suggestion fixed it (thanks).
However, now when I try removing a previous commit of the driver in order to upgrade to this newest one, I get this error:
```
$ sudo make dkms_remove
dkms remove 8812au/5.6.4.2_35491.20191025 --all
Error! The module/version combo: 8812au-5.6.4.2_35491.20191025
is not located in the DKMS tree.
make: *** [Makefile:2311: dkms_remove] Error 3
```
Running `dkms status` on my Fedora 32 machine gives me:
```
$ sudo dkms status
rtl8812au, 5.6.4.2, 5.8.11-200.fc32.x86_64, x86_64: installed
rtl8812au, 5.6.4.2, 5.8.12-200.fc32.x86_64, x86_64: installed
rtl8812au, 5.6.4.2, 5.8.13-200.fc32.x86_64, x86_64: installed
```

Juansbulbisful, 2020-10-14:
Is the shell station busy
...cancell the upgrade.. and all the new incorporated....

b1r63r, 2020-10-16:
The problem with the makefile seems to be in v5.8.5 as well.

kimocoder, 2020-10-16:
branch v5.8.5 is deleted. rtl8814au has been separated into it's own repo. It's standalone from now