aircrack-ng/rtl8812au #750: Simple Stupid Question,But Pls Help :)

GetRektBoy724, 2020-10-13:
I have rtl8821ae and i have kali linux with kernel version 5.8.10 and I want my rtl8821ae can do monitor mode and packet injection.
what driver should i choose if i want packet injection and monitor mode.
should i choose the driver from aircrack-ng github (here) or i choose the driver from astsam ???
pls help :)

jloqfjgk, 2020-11-02:
> I have rtl8821ae

rtl8821**ae** uses PCI-E, so this driver is not for you. This driver is only applicable to the USB one, rtl8821**au**.

GetRektBoy724, 2020-11-02:
oh ok thx for the answer....
@jloqfjgk do you have any alternative driver that support packet injection and monitor mode ???

jloqfjgk, 2020-11-02:
* In [this thread](https://www.linuxquestions.org/questions/slackware-14/problem-trying-to-put-wireless-card-on-monitor-mode-rtl8821ae-slackware64-current-4175605962/), someone says he can put the card into monitor mode in Arch Linux.
* From the list in [hcxdumptool's repo](https://github.com/ZerBea/hcxdumptool), RTL8821AE seems to support monitor mode and full packet injection in kernel >= 5.9.

GetRektBoy724, 2020-11-03:
> * In [this thread](https://www.linuxquestions.org/questions/slackware-14/problem-trying-to-put-wireless-card-on-monitor-mode-rtl8821ae-slackware64-current-4175605962/), someone says he can put the [...]

i think i just need to upgrade to newer kernel 
thx for the answer @jloqfjgk :D

GetRektBoy724, 2020-11-03:
but do i need a specific driver or i just could use the build in driver on my kali linux, @jloqfjgk ??

GetRektBoy724, 2020-11-03:
[image]
i got some evidence that rtl8821ae can run monitor mode and packet injection

jloqfjgk, 2020-11-06:
I think the only Linux driver available for the RTL8821AE is the built-in (in-tree) driver, which means it is already included in the kernel.

GetRektBoy724, 2020-11-09:
ok thx for the answer...you're so helping me @jloqfjgk :)

closed 2021-11-29 by GetRektBoy724