aircrack-ng/rtl8812au #754: Install and Remove scripts are unavailable.

MG-RX93, 2020-10-18:
Why were the install and remove scripts removed?

hifihedgehog, 2020-10-18:
Use make:

https://github.com/aircrack-ng/rtl8812au/issues/747

They embedded the scripts into the makefile. Now, run either make dkms_install or make dkms_remove.

coffee44, 2020-10-20:
I did "make dkms_install" in rtl8812au directory but failed,
wondering I did it in the wrong directory.
Is there anyone who has successfully made it?

kimocoder, 2020-10-20:
Yes.. many times, many

kimocoder, 2020-10-20:
$ git pull
$ make clean && make dkms_install

coffee44, 2020-10-20:
> $ git pull
> $ make clean && make dkms_install

I have successfully installed the driver following your instructions.
Thanks.

MG-RX93, 2020-10-20:
Thanks, this works. Appreciate your help.

closed 2020-10-20 by MG-RX93

Tomaszu97, 2020-10-23:
There is still a part of readme mentioning this:
`$ sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh`
I guess since this file does not exist anymore we can prepend the make command like:
`ARCH=arm make dkms_install` ?

MG-RX93, 2020-10-24:
> There is still a part of readme mentioning this:
> `$ sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh`
> I guess since this file does not exist anymore we can prepend the make command like:
> `ARCH=arm make dkms_install` ?

That should probably solve the issue for people who are expecting to find the install scrips.