aircrack-ng/rtl8812au #755: "Wimplicit-fallthrough" error

CosmicCraftsman, 2020-10-18:
For my version of (Rasbian Stretch Linux RasPi1 4.19.66-v7 armv7l) ("ARM64_RPI") the ("EXTRA_CFLAGS += -Wimplicit-fallthrough=0") line causes a failure with install. Changing "Wimplicit-fallthrough" to "Wno-fallthrough" works and allows it to install with no errors. Don't know if this is a problem on my end, but hopefully this helps someone else or helps locate an underlying issue.

pansila, 2020-10-30:
Same error, same version of raspbina, did you see another error of ‘-mgeneral-regs-only’?

CosmicCraftsman, 2020-10-30:
> Same error, same version of raspbina, did you see another error of ‘-mgeneral-regs-only’?

Unfortunately no.

Thursdays-Child, 2021-02-03:
Same issue on a Raspberry Pi Zero with 4.19.66+ armv6l > ARM_RPI  and same change appears to have permitted the build to continue.

Pansila - I also did not see any -mgeneral-regs-only.

As hazyl mentioned in #759 this change generates new error messages.  In my case: 

~/rtl8812au/core/rtw_ieee80211.c: At top level: 
cc1: warning: unrecognized command line option ‘-Wno-fallthrough=0’

MrRob0-X, 2021-10-30:
> Same issue on a Raspberry Pi Zero with 4.19.66+ armv6l > ARM_RPI and same change appears to have permitted the build to continue.
>
> Pansila - I also did not see any -mgeneral-regs-only.
>
> As hazyl [...]

remove =0 or mark the that cflag

`#EXTRA_CFLAGS += -Wno-fallthrough=0`