aircrack-ng/rtl8812au #756: "No rule to make target modules"

NobodyAnswered, 2020-10-21:
Hi!
I got this message with make && make dkms:
`make && make dkms
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.7.0-kali1-amd64/build M=/home/linux/rtl8812au  modules
make[1]: Entering directory '/usr/lib/modules/5.7.0-kali1-amd64/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.7.0-kali1-amd64/build'
make: *** [Makefile:2246: modules] Error 2
` 

With the make dkms_install method I got this error:
`sudo make dkms_install
mkdir -p /usr/src/8812au-5.6.4.2_35491.20191025
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Error! DKMS tree already contains: 8812au-5.6.4.2_35491.20191025
You cannot add the same module/version combo more than once.
make: *** [Makefile:2305: dkms_install] Error 3
`
If I understand correctly, the second complaint is that I already have the driver, but I don't....

Could someone help me out here?
I would appreciate the help.

Thank you in advance!

gerroon, 2020-10-24:
I have the same error on Debian Testing. I am normally able to compile Gordboy's driver version.

otobrglez, 2020-12-20:
Same problem here. Raspberry Pi V4:

```
sudo make dkms_install
mkdir -p /usr/src/8812au-5.6.4.2_35491.20191025
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Error! DKMS tree already contains: 8812au-5.6.4.2_35491.20191025
You cannot add the same module/version combo more than once.
make: *** [Makefile:2305: dkms_install] Error 3
```

`lsusb` gives

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 005: ID 1546:01a7 U-Blox AG [u-blox 7]
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

kekowar, 2020-12-24:
Same problem with Linux Mint 20  Kernel: 5.10.2-051002-generic x86_64.

`make dkms_install`
mkdir -p /usr/src/8812au-5.6.4.2_35491.20191025
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Error! DKMS tree already contains: 8812au-5.6.4.2_35491.20191025
You cannot add the same module/version combo more than once.
make: *** [Makefile:2305: dkms_install] Error 3

`lsusb`
Bus 003 Device 002: ID 2357:011f TP-Link 802.11ac WLAN Adapter 

Some help here?

debrunet, 2021-06-25:
Remove it from DKMS tree with `sudo dkms remove 8812au/5.6.4.2_35491.20191025 --all`.