aircrack-ng/rtl8812au #759: Compile errors

hazyl, 2020-10-23:
I had the same -Wimplicit-fallthrough=0 not supported error as already reported in #755. The workaround suggested there generated warnings about -Wno-fallthrough being valid in Java but not in C. I tried compiling with  -Wimplicit-fallthrough=0 commented out and with  -Wno-fallthrough, both yielding a first error in os_dep/linux/ioctl_cfg80211.h as follows:

/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/../os_dep/linux/ioctl_cfg80211.h:414:27: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
 #define NL80211_BAND_2GHZ IEEE80211_BAND_2GHZ

Then, several other errors are reported. I am attaching make.log for the commented out  -Wimplicit-fallthrough=0.
Please advise. Thank you/
[make.log](https://github.com/aircrack-ng/rtl8812au/files/5427161/make.log)

hazyl, 2020-10-23:
My kernel is 3.10.0-693.17.1.el7.x86_64 and my gcc is 4.8.5-16.el7_4.1. I am running CentOs 7.4.

danielhollas, 2022-04-21:
@hazyl sorry for reviving this old issue but did you ever manage to make this work? I am trying to compile for the same kernel version and running into the same issue as you describe.