aircrack-ng/rtl8812au #761: SAE WPA3 authentication does not work

haraldrudell, 2020-10-26:
testing 88XXau with b43, Android 10 and newer macOS reveals there is some issue with SAE authentication

as AP, 88XXau and b43 are not WPA3 compliant so macOS and Android 10 will not connect

b43 can connect to b43
88XXau cannot connect to anything, as AP or STA

The SAE authentication response from 88XXau AP causes confirm and disassociate instead of commit like if it is bad password

ValZapod, 2021-09-21:
> SAE authentication response from 88XXau AP causes confirm and disassociate instead of commit like if it is bad password

What happens with a good password then?

emailkanji, 2023-01-14:
@ValZapod That's what happens with a good password.

marcoslimagon, 2023-02-01:
Same behavior here.

154pinkchairs, 2023-02-06:
Doesn't work with neither NetworkMannager nor Connman. And WPA3 support was added approx. 3 years ago. A quick repo search reveals that OSS drivers haven't implemented that apparently.

noraj, 2024-12-23:
- I guess https://github.com/aircrack-ng/rtl8814au is abandoned in favor of https://github.com/aircrack-ng/rtl8812au that works for rtl8814au too.
- Don't buy Alpha AWUS 1900, driver is outdated and buggy, without support for WPA3 and many features
- Alpha claim for "Linux compatibility" yet they provide only Windows and macOS drivers (https://files.alfa.com.tw/?dir=%5B1%5D%20WiFi%20USB%20adapter%2FAWUS1900)
- Other driver discussion about rtl8814au and WPA3 https://github.com/morrownr/8814au/issues/151
- WPA3 works for Alpha AWUS 1900 only on Windows with Windows drivers https://store.rokland.com/blogs/news/wpa3-security-support-now-available-for-awus036ac-awus036ach-awus1900
- RTL8814au not recommended for Linux usage but in-kernel drivers for the rtl8812au and rtl8821/11au chips are NEW as of kernel 6.13 https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Chipsets.md
- Suote from https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapter_Information_for_Linux.md#wpa3-sae-support below

> Most modern out-of-kernel drivers (Realtek) do support WPA3-SAE now but not all. Realtek has discontinued work on most of its WiFi 5 out-of-kernel drivers as of mid-2023, so you must be careful or you may be stuck with a dead-end product.

Hope: 3 days ago [morrownr](https://github.com/morrownr) openned a ticket [8814au in-kernel driver to the Linux Mainline kernel](https://github.com/morrownr/8814au/issues/153) via https://github.com/lwfinger/rtw88