aircrack-ng/rtl8812au #762: DKSM fails to compile with with kernel 5.10-rc1

muncrief, 2020-10-30:
Just a heads up that rtl8812au doesn't compile with kernel 5.10-rc1. I've attached the make log.

[make.log](https://github.com/aircrack-ng/rtl8812au/files/5466480/make.log)

CGarces, 2020-10-30:
I'll take a look 5.10-rc1 has changes that affect many rtl wifi drivers.

muncrief, 2020-10-30:
> I'll take a look 5.10-rc1 has changes that affect many rtl wifi drivers.

Thanks @CGarces. I looked around to see if I could find similar errors, as I've patched the driver before when I could. But in this case the kernel is so new I couldn't find anything. And unfortunately I'm not a kernel developer, just a retired embedded systems designer, so I need an example to guide me.

CGarces, 2020-10-30:
> Thanks @CGarces. I looked around to see if I could find similar errors, as I've patched the driver before when I could. But in this case the kernel is so new I couldn't find anything. And [...]

Just FYI

https://www.zdnet.com/article/linux-5-10-finally-ditches-decades-old-tool-that-caused-security-bugs/
https://github.com/torvalds/linux/commit/f56e65dff6ad52395ef45738799b4fb70ff43376

muncrief, 2020-10-31:
> > > I'll take a look 5.10-rc1 has changes that affect many rtl wifi drivers.
> > 
> > 
> > Thanks @CGarces. I looked around to see if I could find similar errors, as I've patched the driver before when I could. But in this case the kernel is so new I couldn't find anything. And unfortunately I'm not a kernel developer, just a retired embedded systems designer, so I need an example to guide me.
> 
> Just FYI
> 
> https://www.zdnet.com/article/linux-5-10-finally-ditches-decades-old-tool-that-caused-security-bugs/
> [torvalds/linux@f56e65d](https://github.com/torvalds/linux/commit/f56e65dff6ad52395ef45738799b4fb70ff43376)

Yes, I'd read that article, but don't know what to replace set_fs with. And I'm sure whatever replaces it requires different arguments. That's why I was hoping for an example so I could seem like a genius when I'm really not :)

CGarces, 2020-10-31:
Unfortunately, this problem exceed my skills, maybe @kimocoder can help.

muncrief, 2020-10-31:
> Unfortunately, this problem exceed my skills, maybe @kimocoder can help.

I certainly understand. I found quite a few discussions about set_fs going all the way back to 2017, but like I said unfortunately no actual code showing how to replace it. However since its been completely removed now I'm sure something will pop up. If someone here can't fix it now we'll just have to keep our eyes open for a good example elsewhere.

CGarces, 2020-11-06:
@muncrief 
Feel free to copy/paste from

https://github.com/Mange/rtl8192eu-linux-driver/pull/201
Or
https://github.com/lwfinger/rtl8188eu/commit/11bfbc2e3d0abac94a17d3ad054bf0fbd907582b

I'm not sure if `#ifdef set_fs` is enough or need also check the kernel version to maintain compatibility,

Note that I don't have the hardware, the patch was tested only on compilation.

lwfinger, 2020-11-07:
[0001-rtl8812au-Fix-build-errors-from-removal-of-set_fs.txt](https://github.com/aircrack-ng/rtl8812au/files/5505245/0001-rtl8812au-Fix-build-errors-from-removal-of-set_fs.txt)

The attached patch file will fix builds with kernel 5.10.

hanvinke, 2020-11-07:
Thanks lwfinger for the patch. For which branch is the patch? Because I get the errors:

Hunk #1 succeeded at 2191 with fuzz 1 (offset 1406 lines).
Hunk #2 FAILED at 795.
Hunk #3 FAILED at 817.
Hunk #4 FAILED at 852.

muncrief, 2020-11-07:
Thank you @lwfinger, but I'm having the same problem as @hanvinke. Are we using the wrong code? I cloned mine from https://github.com/aircrack-ng/rtl8812au.git. Here's the patch output:

Hunk #1 succeeded at 2191 with fuzz 1 (offset 1406 lines).
Hunk #2 FAILED at 795.
Hunk #3 FAILED at 817.
Hunk #4 FAILED at 852.

I went ahead and tried to compile it anyway since it seems to be more of a generic type patch, but it failed.

lwfinger, 2020-11-07:
v5.2.5

lwfinger, 2020-11-07:
[0001-rtl8812au-Fix-build-errors-from-removal-of-set_fs.txt](https://github.com/aircrack-ng/rtl8812au/files/5505578/0001-rtl8812au-Fix-build-errors-from-removal-of-set_fs.txt)

This patch is for the master branch.

hanvinke, 2020-11-08:
I see, this patch probably meant for your own repo at [https://github.com/lwfinger/rtl8812au](url). Maybe @muncrief can adapt it?

CGarces, 2020-11-08:
Done, #773. 
Note that I don't have the hardware, the patch was tested only on compilation.

muncrief, 2020-11-08:
> I see, this patch probably meant for your own repo at https://github.com/lwfinger/rtl8812au . Maybe @muncrief can adapt it?
> //edit: probably patch is for rtl8188eu : [lwfinger/rtl8188eu@11bfbc2](https://github.com/lwfinger/rtl8188eu/commit/11bfbc2e3d0abac94a17d3ad054bf0fbd907582b)

I adapted this patch for the code at https://github.com/aircrack-ng/rtl8812au.git but unfortunately while it does compile for 5.10, it doesn't work. I tested it with an Edimax AC-1200 model EW-7822UAC and could not connect to a network.

The patch just removes all calls for set_fs, and it appears that something must replace those calls. I took a quick look around for code examples that removed and replaced set_fs but unfortunately I couldn't find any. However if someone can find an example I'll be happy to give it a shot again.

hanvinke, 2020-11-08:
@muncrief  thanks for the info.  Unfortunately I don`t know much too. You need the kernel function access_ok() to check an address against the global variable addr_limit to tell whether to set User- or Kernelspace.

gotti79, 2020-11-09:
I applied the patch with one additional change and for tested it with a noname 8812au device and there I could connect to a AP without problems with a 5.10.0-rc3:
`diff --git a/os_dep/osdep_service.c b/os_dep/osdep_service.c
index 7c6d120..2f6b48d 100644
--- a/os_dep/osdep_service.c
+++ b/os_dep/osdep_service.c
@@ -2164,11 +2164,21 @@ static int writeFile(struct file *fp, char *buf, int len)
 {
        int wlen = 0, sum = 0;
 
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 1, 0))
+       if (!(fp->f_mode & FMODE_CAN_WRITE))
+#else
        if (!fp->f_op || !fp->f_op->write)
+#endif
                return -EPERM;
 
        while (sum < len) {
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0))
+               wlen = kernel_write(fp, buf + sum, len - sum, &fp->f_pos);
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 1, 0))
+               wlen = __vfs_write(fp, buf + sum, len - sum, &fp->f_pos);
+#else
                wlen = fp->f_op->write(fp, buf + sum, len - sum, &fp->f_pos);
+#endif
                if (wlen > 0)
                        sum += wlen;
                else if (0 != wlen)
`
The remove of the set_fs/get_fs should be okay as set_fs/get_fs were used in older kernels in kernel_read and kernel_write functions which were changed to not need it anymore. And here kernel_read and kernel_write will be get used this should work as before. According to [https://github.com/mkubecek/vmware-host-modules/issues/76](url) this should work as long as you don't use eventfd.

CGarces, 2020-11-09:
Added @gotti79  code to #773
The code is provided As is, without testing, please confirm in case it works fine.

hanvinke, 2020-11-09:
Unfortunately the build still fails for me. But thanks for all the people involved!

[builderrors.txt](https://github.com/aircrack-ng/rtl8812au/files/5512770/builderrors.txt)

CGarces, 2020-11-09:
@hanvinke, the build compiles fine, you are compiling a branch without patching.
Please test the branch of #773
https://github.com/CGarces/rtl8812au/tree/fix_kernel_5.10

muncrief, 2020-11-09:
> Unfortunately the build still fails for me. But thanks for all the people involved!
> 
> [builderrors.txt](https://github.com/aircrack-ng/rtl8812au/files/5512770/builderrors.txt)

I created a patch for the code at https://github.com/aircrack-ng/rtl8812au.git. It's a modified combination of the patches by @lwfinger and @gotti79. Thanks to both of them for their work.

I do want to warn that this patch does not at all leave me with a comfortable feeling, and I have no idea what the implications of it are. However I know many people use the main code repository so I've done the best I can. I tested it on Arch with kernels 5.9.6 and 5.10.0-rc3, using an Edimax AC-1200 model EW-7822UAC USB WiFi adapter. 
[rtl8812au-5.10.patch.txt](https://github.com/aircrack-ng/rtl8812au/files/5513301/rtl8812au-5.10.patch.txt)

hanvinke, 2020-11-09:
@CGarces 
I don´t think i used the wrong branch. 
git clone --single-branch --branch fix_kernel_5.10 https://github.com/CGarces/rtl8812au.git

muncrief, 2020-11-09:
> @CGarces
> I don´t think i used the wrong branch.
> git clone --single-branch --branch fix_kernel_5.10 https://github.com/CGarces/rtl8812au.git

It's not necessarily a wrong branch, it's just that many people are led to https://github.com/aircrack-ng/rtl8812au.git, at least when searching for Edimax drivers. In any case I'd recommend people use the branch you're suggesting, as like I said I have no idea what I'm doing :)

CGarces, 2020-11-09:
> @CGarces
> I don´t think i used the wrong branch.
> git clone --single-branch --branch fix_kernel_5.10 https://github.com/CGarces/rtl8812au.git

Compilation over last mainline (5.10-rc3) works fine, the build process is automated by Travis CI
Feel free to check the logs at:

https://travis-ci.com/github/CGarces/rtl8812au/jobs/431433531

Note that build is failing right now, but not is related to this issue. is a weird [environment problem with the ubuntu kernel](https://www.spinics.net/lists/kernel/msg3716600.html), but the compilation works fine.

My tests are with amd64 kernel. `set_fs` not is defined on that kernel.
Your build error log only is posible if `set_fs` is defined or if you are using a branch without #773 

¿Maybe you are testing with **arm64** or **arm**? `set_fs` is defined on that architectures.
https://github.com/torvalds/linux/commit/f56e65dff6ad52395ef45738799b4fb70ff43376#diff-95d57ff480709dc78ab50b40e311474847565550856db04681e1ba3c598af824R123

muncrief, 2020-11-10:
> > @CGarces
> > I don´t think i used the wrong branch.
> > git clone --single-branch --branch fix_kernel_5.10 https://github.com/CGarces/rtl8812au.git
> 
> Compilation over last mainline (5.10-rc3) works fine, the build process is automated by Travis CI
> Feel free to check the logs at:
> 
> https://travis-ci.com/github/CGarces/rtl8812au/jobs/431433531
> 
> Note that build is failing right now, but not is related to this issue. is a weird [environment problem with the ubuntu kernel](https://www.spinics.net/lists/kernel/msg3716600.html), but the compilation works fine.
> 
> My tests are with amd64 kernel. `set_fs` not is defined on that kernel.
> Your build error log only is posible if `set_fs` is defined or if you are using a branch without #773
> 
> ¿Maybe you are testing with **arm64** or **arm**? `set_fs` is defined on that architectures.
> [torvalds/linux@f56e65d#diff-95d57ff480709dc78ab50b40e311474847565550856db04681e1ba3c598af824R123](https://github.com/torvalds/linux/commit/f56e65dff6ad52395ef45738799b4fb70ff43376#diff-95d57ff480709dc78ab50b40e311474847565550856db04681e1ba3c598af824R123)

I used "git clone --single-branch --branch fix_kernel_5.10 https://github.com/CGarces/rtl8812au.git" and tested the the code. It compiles and runs on Arch kernels 5.9.6 and 5.10.0-rc3 with my Edimax AC-1200 model EW-7822UAC USB WiFi adapter. Good job!

Just to be clear for other users, the CGarces code is already patched, so all you need to do is compile it with "sudo make dkms_install"

Afterwords you can install it on other kernels with "sudo dkms install 8812au/5.6.4.2_35491.20191025 -k <kernel name>"

hanvinke, 2020-11-10:
Hi, I tested again. As it turns out the kernel image I was using from https://kernel.ubuntu.com/~kernel-ppa/mainline/daily/current/ was missing some kernel features. Journalctl showed AppArmor complaining about `some kernel features are missing: dbus, network`. This apparently is `normal` for the daily kernel builds.  Of course I tried again with https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.10-rc3/ 
This time no errors building. Thanks to all!

hanvinke, 2020-11-15:
This environment problem with the kernel in Ubuntu can be solved by renaming **/usr/src/linux-headers-5.10.0-051000rc3/scripts/module.lds.S** to **module.lds**. Then remove the last line (29).

alxchk, 2020-12-14:
Tested on Alpha AC1200 / 5.10.0 -- looks good.

PR #773 (2020-12-22): Fix kernel 5.10-rc1
Fix compilation issues on kernel 5.10
The change is needed due this change.
https://github.com/torvalds/linux/commit/f56e65dff6ad52395ef45738799b4fb70ff43376

Patch for 5.1 has been refactored, get_ds() is an alias for KERNEL_DS

Note that #760 is need to test CI build. 
Please @kimocoder consider migrate to travis.com
https://docs.travis-ci.com/user/migrate/open-source-repository-migration#migrating-a-repository

Fix #762
closed 2020-12-22 by kimocoder