aircrack-ng/rtl8812au #763: dkms not working

hackingking466, 2020-10-31:
`root@kali:~# apt-get install dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dkms is already the newest version (2.8.3-4).
The following packages were automatically installed and are no longer required:
  dwarfdump killerbee libarmadillo9 libcdio18
  libcfitsio8 libdistorm3-3 libdwarf1
  libgtksourceview2.0-0 libgtksourceview2.0-common
  libilmbase24 libjsoncpp1 libmdb2 libmdbsql2
  libmpdec2 libobjc-9-dev libopenexr24 libpgm-5.2-0
  libpoppler82 libprotobuf22 libsnmp35
  libsrt1-gnutls libtsk13 libx264-155 libx264-159
  libyara3 linux-headers-5.8.0-kali2-amd64
  linux-headers-5.8.0-kali2-common openjdk-8-jre
  python-cairo python-crypto python-dbus
  python-distorm3 python-enchant python-et-xmlfile
  python-gi python-gobject-2 python-gtk2
  python-gtksourceview2 python-jdcal python-libxml2
  python-libxslt1 python-lxml python-numpy
  python-olefile python-openpyxl python-pil
  python-serial python-usb python-yara
  python3-flask-restless python3-grequests
  python3-mimeparse python3-mimerender ruby-arel
  ruby-chunky-png ruby-rqrcode volatility
  volatility-tools
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@kali:~# make dkms_install
make: *** No rule to make target 'dkms_install'.  Stop.
root@kali:~# make && make install
make: *** No targets specified and no makefile found.  Stop.
root@kali:~# ./dkms-install.sh
-bash: ./dkms-install.sh: No such file or directory
root@kali:~# apt install realtek-rtl88xxau-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
realtek-rtl88xxau-dkms is already the newest version (5.6.4.2~git20200916-0kali1).
The following packages were automatically installed and are no longer required:
  dwarfdump killerbee libarmadillo9 libcdio18 libcfitsio8 libdistorm3-3
  libdwarf1 libgtksourceview2.0-0 libgtksourceview2.0-common libilmbase24
  libjsoncpp1 libmdb2 libmdbsql2 libmpdec2 libobjc-9-dev libopenexr24
  libpgm-5.2-0 libpoppler82 libprotobuf22 libsnmp35 libsrt1-gnutls libtsk13
  libx264-155 libx264-159 libyara3 linux-headers-5.8.0-kali2-amd64
  linux-headers-5.8.0-kali2-common openjdk-8-jre python-cairo python-crypto
  python-dbus python-distorm3 python-enchant python-et-xmlfile python-gi
  python-gobject-2 python-gtk2 python-gtksourceview2 python-jdcal
  python-libxml2 python-libxslt1 python-lxml python-numpy python-olefile
  python-openpyxl python-pil python-serial python-usb python-yara
  python3-flask-restless python3-grequests python3-mimeparse
  python3-mimerender ruby-arel ruby-chunky-png ruby-rqrcode volatility
  volatility-tools
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@kali:~# cd Downloads
root@kali:~/Downloads# cd rt
root@kali:~/Downloads/rt# sudo ./dkms-install.sh
sudo: ./dkms-install.sh: command not found
root@kali:~/Downloads/rt# `
i cant install it please help!!

4spuds, 2020-10-31:
See issue #747 titled "Change README to instruct to install with DKMS via Makefile"
...
Try:
$ sudo make dkms_install

hackingking466, 2020-11-01:
root@kali:~/Downloads/rt# make dkms_install
make: *** No rule to make target 'dkms_install'.  Stop.

``
what should i do thats the result i get

hackingking466, 2020-11-01:
i downloaded it but my AWUS1900 dont work i know it works because i tried it on a windoes pc and it worked so anyone can tell me why it still doesnt work?

cedricbambooza, 2020-12-31:
AWUS1900 has the Realtek RTL8814U Chipset (Reference https://www.wifi-shop24.com/alfa-networks-awus1900-wifi-usb-adapter), so you need the other package as described on the "Code" Site.

cedricbambooza, 2021-01-07:
consider closing the issue or reopen on the site for the right driver :) 

https://github.com/aircrack-ng/rtl8814au