aircrack-ng/rtl8812au #765: Kali 5.7.0 - Awus036ACH - rtl8812au - Deauth problem

Craftonix, 2020-11-03:
Hello,
I'm running Kali version :
Linux kali 5.7.0-kali1-amd64 #1 SMP Debian 5.7.6-1kali2 (2020-07-01) x86_64 GNU/Linux
under virtualbox VM and also tested on a live usb key

Installed the driver with make & make install , (also test with dkms), i use the branch v5.6.4.2

I put the usb wifi in monitor mode using the following trick ;
ip link set wlan0 down
iw dev wlan0 set type monitor
ip link set wlan0 up
iw wlan0 set txpower fixed 3000

then i run the classic airodump-ng / aireplay-ng --deauth
but i don't get any handshake, is the airodump-ng not seeing packet or deauth not working ?

With another wifi usb key using rt73usb driver, it works like a charm, get the handshake of my own routeur in a minute

Any idea ? Kali too recent ?

Thanks for you kind support.

hackingking466, 2020-11-03:
can u write how u downloaded the drivers for the alfa? i have AWUS1900 and i can't get it up and running

Craftonix, 2020-11-03:
> 
> 
> can u write how u downloaded the drivers for the alfa? i have AWUS1900 and i can't get it up and running

The AWUS1900 driver is not anymore in this git repo.
go and check the instruction here :
https://github.com/aircrack-ng/rtl8814au

Craftonix, 2020-11-06:
I found the problem :
The commands to set in monitor mode using iw seems not to be correct or incomplete.
i added this to my monitor enable script
ifconfig wlan0 down
iwconfig wlan0 mode monitor
ifconfig wlan0 up

and now the injection/deauth work better, i now capture handshake correctly.
I will dig it now what the difference between iw and iwconfig

closed 2020-11-06 by Craftonix