aircrack-ng/rtl8812au #774: Can't compile driver on Debian (kernel v. 5.9.0-1 amd64)

developerofshadows, 2020-11-10:
I'm typing make in console, and get this: 
`#make ARCH=x86_64 SUBARCH=x86_64 REAL_CC=/clang CLANG_TRIPLE=aarch64-linux-gnu- CROSS_COMPILE= -C /lib/modules/5.9.0-1-amd64/build M=/home/work/gits/rtl8812au O="" modules
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.9.0-1-amd64/build M=/home/work/gits/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.9.0-1-amd64'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[2]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:718: include/config/auto.conf] Error 1
make[1]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.9.0-1-amd64'
make: *** [Makefile:2312: modules] Error 2
`

yelsanya, 2021-01-10:
same here with 5.9.0-kali5-amd64

CGarces, 2021-01-11:
Hi.

The log said "Run 'make oldconfig && make prepare' on kernel src to fix it."

Have you done it?

Looks like you are trying to made a x86_64 build on a arm64 device?