aircrack-ng/rtl8812au #775: Kali 2020.3  rtl8812au  problem

LuckyFishGeek, 2020-11-11:
I can attack 2.4G wifi after using the following steps
But unable to attack 5G wifi
Has any friend encountered the same problem as me?
The chip is rtl8812au

In addition, neither Fluxion nor wifiphisher can use the network card of this chip.

vitorinux, 2020-11-22:
Check this driver, for me works on slackware kernel 5.4.73

https://github.com/morrownr/8812au

version: 8812au = v5.9.3.2_37279.20201012
vermagic: 5.4.73-slackware SMP

LuckyFishGeek, 2020-11-22:
> Check this driver, for me works on slackware kernel 5.4.73
> 
> https://github.com/morrownr/8812au
> 
> version: 8812au = v5.9.3.2_37279.20201012
> vermagic: 5.4.73-slackware SMP

thanks  I‘ll try it

closed 2020-12-05 by LuckyFishGeek