aircrack-ng/rtl8812au #778: Radio scans introduce lag spikes

tdragan, 2020-11-23:
environment: 
  ubuntu,popOS 20.04, 20.10 
kernel: 
  5.4, 5.8 

issue: 
In what seems random intervals, there are lag spikes taking about 9 seconds which block network activity. Following closely ping to any site it can be seen that spikes occur consistently. When Network Manager is disabled it doesn't happen, because scan are not initiated. If I explicitly run "wpa_cli scan" I can reproduce the problem even if NM is turned off, so its not NM's fault. So I guess its wpa_supplicant/driver issue. My other devices (android phones and windows 10) have clean ping so I don't think its router related.

tdragan, 2020-12-04:
Should it block all traffic? It's like 9 seconds of disconnect if I play some multiplayer game I will be kicked due to timeout. Doesn't seem normal to me

raplin, 2020-12-04:
Yes; literally the radio in the wifi dongle has to retune one at a time to all the other frequencies (separately on 2.4 and 5.8), then send a Probe Request, then listen for responses, and repeat. During this time it is simply not tuned to the frequencies of the AP it's connected to and can't do any comms.

raplin, 2020-12-04:
solution is just to not have card scan when it's associated, this is a wpa_supplicant setting

tdragan, 2020-12-04:
I will try to find spa setting. I have set fixed BSSID but it doesn't help (as some source indicate). Very informative - thanks.

mrbaguettovic, 2025-05-28:
Facing the same problem, @tdragan what setting did you use?