aircrack-ng/rtl8812au #780: Doesn't work on Suse

martinmei71, 2020-11-28:
I'm a user of openSUSE Tumblewed and this driver doesn't install i search other options but none works.

arnoldix, 2021-06-27:
I had the same issue when updating from Leap 15.2 to Leap 15.3.

The make command ended with this messages:
/root/realtek-rtl88xxau-dkms/os_dep/linux/ioctl_cfg80211.c:10191:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
/root/realtek-rtl88xxau-dkms/os_dep/linux/ioctl_cfg80211.c:10191:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/realtek-rtl88xxau-dkms/os_dep/linux/ioctl_cfg80211.c:10191:25: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-5.3.18-59.5/scripts/Makefile.build:288: /root/realtek-rtl88xxau-dkms/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** [/usr/src/linux-5.3.18-59.5/Makefile:1675: _module_/root/realtek-rtl88xxau-dkms] Error 2
make[1]: *** [../../../linux-5.3.18-59.5/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-5.3.18-59.5-obj/x86_64/default'
make: *** [Makefile:2244: modules] Error 2

If you use the same version as I did you will find in the file os_dep/linux/ioctl_cfg80211.c the following line twice
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0))
Replace this by
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,3,0))
and re-run the make command. This worked on my system.

(remark: Both Leap 15.2 and Leap 15.3 use currently a kernel version 5.3.18 and the make command runs in Leap 15.2 without the described modification. I have no idea if the modification of struct cfg80211_ops in this a 5.3.x kernel is a SUSE topic or if this a modification of the official vanilla kernel.)

CGarces, 2021-06-28:
Please check issue #861

Patch proposed at https://github.com/aircrack-ng/rtl8812au/issues/861#issuecomment-859794427