aircrack-ng/rtl8812au #784: Kali 2020.4 rtl8812au Virtual interface problem

LuckyFishGeek, 2020-12-05:
kali kernel:5.9.3.2

Both Airgeddon and fluxion cannot use the virtual interface of the network card

Drivers currently tested

https://github.com/morrownr/8812au

https://github.com/gordboy/rtl8812au-5.9.3.2

apt install realtek-rtl88xxau-dkms

All don’t support virtual interface mode. Has any friend solved this problem?

Metal8, 2020-12-05:
Having the same problem with Kali 5.9.6 Kernel.

desegura, 2020-12-06:
Same problem after updating Kali

usama7628674, 2020-12-17:
@desegura @Metal8 @LuckyFishGeek This is not a driver issue.Realtek didn't write driver for `mac80211` so VIF is not supported.You need to get a second adapter to use for AP or buy a compatible adapter that supports VIF.I recommend Panda PAU09.

Medullitus, 2022-02-02:
Help please. I have the same chipset please help!!!

Auguss82, 2022-07-10:
Seeing this is just not me. This was previously working before recent upgrade and dist-upgrade on kali distro. previously it was working.

benlajunet, 2025-03-31:
im still reading, waiting to RTL8812AU Can run fluxion with no problema