aircrack-ng/rtl8812au #793: No association when using 5GHz Wi-Fi network (RTL8811AU)

agners, 2020-12-16:
I am trying to use this driver with a Vanilla Linux 5.10.1 kernel. It compiles and loads fine, however, when trying to associate using 5GHz network it fails to associate:

```
nmcli device wifi connect "My 5GHz SSID" password "mycorrectpw"
Error: Connection activation failed: (7) Secrets were required, but not provided.
```

```
Dec 16 16:22:43 homeassistant wpa_supplicant[389]: wlan0: Trying to associate with 04:ab:cd:ef:ef:ef (SSID='My 5GHz SSID' freq=5220 MHz)
Dec 16 16:22:43 homeassistant wpa_supplicant[389]: nl80211: kernel reports: Authentication algorithm number required
Dec 16 16:22:43 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5220 ht_enabled=1 ch_offset=1 ch_width=40 MHz cf1=5230 cf2=0
Dec 16 16:22:43 homeassistant NetworkManager[369]: <info>  [1608135763.2104] device (wlan0): supplicant interface state: scanning -> associating
Dec 16 16:22:43 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-ASSOC-REJECT status_code=54
Dec 16 16:22:43 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
Dec 16 16:22:43 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Dec 16 16:22:43 homeassistant NetworkManager[369]: <info>  [1608135763.3480] device (wlan0): supplicant interface state: associating -> disconnected
Dec 16 16:22:43 homeassistant NetworkManager[369]: <warn>  [1608135763.3482] sup-iface[0x3206b910,wlan0]: connection disconnected (reason 54)
Dec 16 16:22:44 homeassistant NetworkManager[369]: <info>  [1608135764.3542] device (wlan0): supplicant interface state: disconnected -> scanning
Dec 16 16:22:48 homeassistant wpa_supplicant[389]: wlan0: Trying to associate with 04:ab:cd:ef:ef:ef (SSID='My 5GHz SSID' freq=5220 MHz)
Dec 16 16:22:48 homeassistant wpa_supplicant[389]: nl80211: kernel reports: Authentication algorithm number required
Dec 16 16:22:48 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5220 ht_enabled=1 ch_offset=1 ch_width=40 MHz cf1=5230 cf2=0
Dec 16 16:22:48 homeassistant NetworkManager[369]: <info>  [1608135768.2197] device (wlan0): supplicant interface state: scanning -> associating
Dec 16 16:22:48 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-ASSOC-REJECT status_code=54
Dec 16 16:22:48 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="My 5GHz SSID" auth_failures=1 duration=10 reason=CONN_FAILED
Dec 16 16:22:48 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
Dec 16 16:22:48 homeassistant wpa_supplicant[389]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
```

I tried with 80MHz/40MHz as well as 20MHz channel width. I am using this `/etc/NetworkManager/NetworkManager.conf`:

```
[device]
wifi.scan-rand-mac-address=no
```

2.4GHz seems to work.

The USB device PID/VID is `0bda:a811`.

agners, 2020-12-16:
This looks like the same issue as seen in #664.

agners, 2020-12-16:
Forgot to mention: that is using default branch v5.6.4.2 as of today (e9fbf5c).

azratul, 2021-03-29:
Same here. TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano] works fine with 2.4 Ghz networks, but cannot see 5Ghz networks and kernel error is the same already reported. Kernel reports: Authentication algorithm number required
Arch linux kernel 5.11.8-arch1-1

ccxex29, 2021-04-26:
Same here. It actually works sometimes. But most of the time it just got stuck in the Association phase for a long time until it is timeout.  

But when it does, the 5GHz connection would be in Configuration state instantaneously. I usually try to turn the wifi interface off and on again from connman or from the terminal using `ip link set wlp0s20f0u4 down` and `ip link set wlp0s20f0u4 up`.

I use connman, wpa_supplicant on Artix linux-tkg 5.11.16-149-tkg-upds with TP-LINK Archer T2U Nano WiFi.