aircrack-ng/rtl8812au #794: Deauth not work (RTL8812AU)

vincenzogianfelice, 2020-12-17:
Hi. I did a couple of tests and i noticed that the deauth (5ghz) doesn't work. These are the commands (in sequence that I ran):

```
iwconfig wlan1 mode monitor
iwconfig wlan1 channel 36
aireplay-ng -0 0 -a XX.XX.XX... wlan1
```

My configuration is:

Debian 9 (x64)
Kernel 4.9.0
aircrack-ng 1.2

MAGADE007, 2020-12-20:
Hi, first for my english but it is not my native language.
I'm running all fresh kali 2020.4 directly hosted on my pc (not vm).
I'm alpha awsu036ach user. Same issue
more than 3 days to try to work aireplay deauth.
I succes only one time to deauth, impossible to succes again? it seem to send well deauth packet to the mac device i want to deauth but de device continue to get internet and this also with 0 (continious dos attaque mode)
Somebody have fix this issue? Thanks

dkmc360, 2020-12-24:
I am having the same issue. I am unable to deauth.

MAGADE007, 2020-12-24:
deauth on 2.4 and 5ghz doesn't work. Seems to work but not. I also remove all physical wifi card on my laptop to get only one wifi card installed on motherboard but no succes

MAGADE007, 2020-12-24:
I have tried lot of different driver (from aircrack, from asman, rtl8812au, rtl88xxau etc. 
Have tried also to remove my internal wifi card from the motherboard of my laptop to get only one wifi adaptation.doesn't work
Tried on vm also but not succes.
different way to command line with airman-ng or ifconfig ifconfig, ip link set etc. nothings work

vincenzogianfelice, 2020-12-27:
I ordered another 5ghz alfa card which should arrive in days (awus1900) and it has another (slightly improved) chipset, **rtl8814au**.

We hope well.

Happy holidays to all.

dkmc360, 2020-12-27:
Got mine to work via kali by uninstalling the drivers from GitHub and installing the drivers from kali.

kimocoder, 2020-12-27:
I have new driver version available shortly, just hold on a bit

kimocoder, 2020-12-28:
Ok guys, try this one.. it's still under development/testing. New 8812AU driver with newer 8821AU HAL (8811) included.

https://github.com/kimocoder/rtl8812au-5.9.3.2

Please report back if deauth works!

1yura, 2020-12-28:
https://github.com/kimocoder/rtl8812au-5.9.3.2
deauth works for both rtl8812au and rtl8811au, but on the 5GHz 
```aireplay-ng --deauth``` 
works only with the -D option. Otherwise "No such BSSID available."

kimocoder, 2020-12-28:
Try be experimental and move over to the newer branch v5.7.0. newer HAL/phydm for both 8812au and 8821au 👍 

Presse report back if that solves it

1yura, 2020-12-29:
v5.7.0 - deauth works now. But it seems only 36 .. 48 channels are enabled.
v5.9.3.2 - deauth works. There is only one AP that aireplay-ng does not see without -D option. It is on the 149 channel, and i can not test it with v5.7.0: "kernel reports: Channel is disabled".

kimocoder, 2020-12-29:
Haven't unlocked all channels yet, beware of the issue 👍

vincenzogianfelice, 2020-12-30:
Hello.  I think I understand the problem, but I need someone to try.  I currently have an awus1900 and I recently returned the awus36ach, so I can't try.  **Have you tried setting the country with** `iw reg set`?  try deauth by setting one of the following countries each time:

`iw reg set US
iw reg set BO
iw reg set IT
`

let me know, because if it works I will send this back and take the awus036ach again.  Also on the latter I encountered problems() and I hope that @kimocoder can explain to me why.

Thanks

MAGADE007, 2021-01-01:
Hi, todays, I tested my alpha AWSU036ACH and my PAU09 wifi adaptation. After a update, upgrade, dist-upgrade it seem to work well for both wifi card (Deauth, handshake, 2.4 and 5 ghz). Thanks

sandman6578, 2021-01-09:
For me deauth when using
 git clone this repo 
also dont works.

FYI its the same on my awus1900 with rtl8814au

ExpertInLinux, 2021-01-12:
Did @vincenzogianfelice or anyone get the awus1900 working with deauth? I've got the same problem and tried all drivers..

vincenzogianfelice, 2021-01-12:
@ExpertInLinux Does not work. I have returned the card in question as I purchased it for this purpose. Unfortunately there is still some time to mature on these drivers ... it will take time before they work in the best way. For the moment I have sent the antenna back.

ExpertInLinux, 2021-01-12:
@vincenzogianfelice thanks for your reaction, I just bought it. Monitor mode is great and it's reach is good. Really don't know what I'm going to do next with it.. maybe stick to my archer t2uh for a while.

snoopspy, 2021-01-30:
FYI, https://github.com/aircrack-ng/rtl8812au/issues/819

astsam2, 2021-01-31:
Hi snoopspy! May you confirm the patch in the #819 is working for you? I am trying  to work around issue with injection in the v5.6.4.2.

snoopspy, 2021-02-01:
Hi @astsam2. #819 work fine for me.
https://youtu.be/jq2-BU5C_PI

ExpertInLinux, 2021-02-01:
@snoopspy pspy Does this work with the RTL8814AU too?

usuarionuevor, 2021-02-11:
> Ok guys, try this one.. it's still under development/testing. New 8812AU driver with newer 8821AU HAL (8811) included.
> 
> https://github.com/kimocoder/rtl8812au-5.9.3.2
> 
> Please report back if deauth works!

it works!! thanks (rtl8812au)

DeityOfChaos, 2021-02-17:
I am having this issue, on raspberry pi and kali. Deauth not working, 1m distance to client and AP  (rtl8812au)

JBian-stack, 2021-02-19:
Thank you for this. Deauth for my RTL8812AU product works on 2.4 Ghz but not exactly on 5 Ghz. I see some packet transfer, but mostly  0's.

0 | 0
0 | 0
1 | 2
0 | 1
0 | 0
0 | 0

Etc. Etc.

I leave it running for 1/2 hour but the deauth attack is not successful. I have tried two 5 Ghz routers and various devices. I know there are other reasons why a deauth can fail. Is the low packet rate above a sign of any direction I should look?

DeityOfChaos, 2021-02-20:
> Thank you for this. Deauth for my RTL8812AU product works on 2.4 Ghz but not exactly on 5 Ghz. I see some packet transfer, but mostly 0's.
> 
> 0 | 0
> 0 | 0
> 1 | 2
> 0 | 1
> 0 | 0
> 0 | 0
> 
> Etc. Etc.
> 
> On 2.4 Ghz the numbers are much higher and the deauth is quickly successful.
> 
> On 5 GHz leave it running for 1/2 hour but the deauth attack is not successful. I have tried two 5 Ghz routers and various devices. I know there are other reasons why a deauth can fail. Is the low packet rate above a sign of any direction I should look?

I used this git https://github.com/morrownr/8812au and I was able to deauth on 5Ghz, only had to add the -D option.

JBian-stack, 2021-02-23:
Thanks DeityOfChaos

This worked great:
https://github.com/morrownr/8812au

Dr-Insanity, 2022-10-17:
Kimocoder, you're doing a great job for us. I thank you for maintaining this. I thought it had to be said. :D

mkarne1, 2023-04-25:
Hi there!

I can't deauthenticate any client from any Wifi. I installed the [latest driver](https://github.com/aircrack-ng/rtl8812au) with the following commands: 

```
sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-`uname -r`
reboot
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl
make dkms_install  
```

Afterwards, I ran:

```
airmon-ng check kill
ip link set wlan0 down
iw dev wlan0 set type monitor
ip link set wlan0 up
airodump-ng --bssid D4:3D:F3:74:F3:91 --channel 5 --write /tmp/test wlan0
```

And in a new terminal:

```
aireplay-ng --deauth 1000 -a D4:3D:XX:XX:XX:91 -c 14:7D:XX:XX:XX:44 wlan0          
19:46:04  Waiting for beacon frame (BSSID: D4:3D:F3:74:F3:91) on channel 5
19:46:05  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [31|73 ACKs]
19:46:05  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [74|89 ACKs]
19:46:06  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [136|118 ACKs]
19:46:06  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 2|63 ACKs]
19:46:07  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 0|63 ACKs]
19:46:07  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 1|64 ACKs]
19:46:08  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [14|65 ACKs]
19:46:08  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [89|89 ACKs]
19:46:09  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [74|88 ACKs]
19:46:09  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [128|119 ACKs]
```

It just keeps sending deauth packets but the client isn't deauthenticating. 

Any help would be really appreciated!

@kimocoder do you have any idea what I'm doing wrong? 

Thanks in advance!