aircrack-ng/rtl8812au #797: Errors compiling for openwrt for onion omega2

abiuan, 2020-12-22:
Hello,
I'm trying to cross compile the driver for openwrt distro running on onion omega2. I'm using version 5.7.0 of the driver. The kernel of openwrt is version 4.4.81. I'm trying to build as OpenWrt package. I installed the package, ran "make menuconfig".
When I start the compilation using "make" I get the following errors.
error "Must be LITTLE or BIG Endian" means that linux-ramips_mt76x8 is not a supported CPU?

Best regards

make[4]: Entering directory '/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/linux-4.14.81'
  CC [M]  /root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/core/rtw_cmd.o
In file included from /root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/drv_types.h:27:0,
                 from /root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/core/rtw_cmd.c:17:
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/rtw_byteorder.h:30:5: error: #error "Must be LITTLE/BIG Endian Host"
  #  error "Must be LITTLE/BIG Endian Host"
     ^~~~~
In file included from /root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/drv_types.h:96:0,
                 from /root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/core/rtw_cmd.c:17:
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/rtw_mlme_ext.h:1229:4: error: #error "Must be LITTLE or BIG Endian"
 #  error "Must be LITTLE or BIG Endian"
    ^~~~~
In file included from /root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_precomp.h:92:0,
                 from /root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/hal_data.h:20,
                 from /root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/core/rtw_cmd.c:18:
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_phystatus.h:61:6: warning: "ODM_ENDIAN_TYPE" is not defined, evaluates to 0 [-Wundef]
 #if (ODM_ENDIAN_TYPE == ODM_ENDIAN_LITTLE)
      ^~~~~~~~~~~~~~~
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_phystatus.h:87:6: warning: "ODM_ENDIAN_TYPE" is not defined, evaluates to 0 [-Wundef]
 #if (ODM_ENDIAN_TYPE == ODM_ENDIAN_LITTLE)
      ^~~~~~~~~~~~~~~
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_phystatus.h:110:6: warning: "ODM_ENDIAN_TYPE" is not defined, evaluates to 0 [-Wundef]
 #if (ODM_ENDIAN_TYPE == ODM_ENDIAN_LITTLE)
      ^~~~~~~~~~~~~~~
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_phystatus.h:124:6: warning: "ODM_ENDIAN_TYPE" is not defined, evaluates to 0 [-Wundef]
 #if (ODM_ENDIAN_TYPE == ODM_ENDIAN_LITTLE)
      ^~~~~~~~~~~~~~~
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_phystatus.h:133:6: warning: "ODM_ENDIAN_TYPE" is not defined, evaluates to 0 [-Wundef]
 #if (ODM_ENDIAN_TYPE == ODM_ENDIAN_LITTLE)
      ^~~~~~~~~~~~~~~
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_phystatus.h:152:6: warning: "ODM_ENDIAN_TYPE" is not defined, evaluates to 0 [-Wundef]
 #if (ODM_ENDIAN_TYPE == ODM_ENDIAN_LITTLE)
      ^~~~~~~~~~~~~~~
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_phystatus.h:170:6: warning: "ODM_ENDIAN_TYPE" is not defined, evaluates to 0 [-Wundef]
 #if (ODM_ENDIAN_TYPE == ODM_ENDIAN_LITTLE)
      ^~~~~~~~~~~~~~~
/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/include/../hal/phydm/phydm_phystatus.h:181:6: warning: "ODM_ENDIAN_TYPE" is not defined, evaluates to 0 [-Wundef]
 #if (ODM_ENDIAN_TYPE == ODM_ENDIAN_LITTLE)
      ^~~~~~~~~~~~~~~
scripts/Makefile.build:328: recipe for target '/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/core/rtw_cmd.o' failed
make[5]: *** [/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/core/rtw_cmd.o] Error 1
Makefile:1525: recipe for target '_module_/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de' failed
make[4]: *** [_module_/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de] Error 2
make[4]: Leaving directory '/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/linux-4.14.81'
Makefile:64: recipe for target '/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/.built' failed
make[3]: *** [/root/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8812au-ac-cc4c64de/.built] Error 2
make[3]: Leaving directory '/root/rtl8812au/openwrt/rtl8812au-ac'
Command exited with non-zero status 2
time: package/feeds/rtl8812au/rtl8812au-ac/compile#1.34#0.32#1.50
package/Makefile:107: recipe for target 'package/feeds/rtl8812au/rtl8812au-ac/compile' failed
make[2]: *** [package/feeds/rtl8812au/rtl8812au-ac/compile] Error 2
make[2]: Leaving directory '/root/source'
package/Makefile:103: recipe for target '/root/source/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/root/source/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/source'
/root/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

lucize, 2020-12-27:
I'm also trying but from what I see

```
ifeq ($(CONFIG_BIG_ENDIAN), y)
NOSTDINC_FLAGS+= -DCONFIG_BIG_ENDIAN
endif

ifeq ($(CONFIG_LITTLE_ENDIAN), y)
NOSTDINC_FLAGS+= -DCONFIG_LITTLE_ENDIAN
endif

```

is not evaluated correctly

lucize, 2020-12-28:
@abiuan 
can you try this for your case?

```
define Build/Compile
        +$(MAKE) $(PKG_JOBS) -C "$(LINUX_DIR)" \
                $(KERNEL_MAKE_FLAGS) \
                $(PKG_MAKE_FLAGS) \
                M="$(PKG_BUILD_DIR)" \
                $(if $(CONFIG_BIG_ENDIAN),NOSTDINC_FLAGS="$(NOSTDINC_FLAGS) -DCONFIG_BIG_ENDIAN", \
                NOSTDINC_FLAGS="$(NOSTDINC_FLAGS) -DCONFIG_LITTLE_ENDIAN") \
                modules
endef

```

abiuan, 2020-12-28:
Sorry Lucian,

what file are you talking about? "./Makefile"?

lucize, 2020-12-29:
for Openwrt master and 5.7.0 branch you will need a patch like this:

```
--- a/os_dep/linux/ioctl_cfg80211.c     2020-12-29 01:33:09.731405872 +0200
+++ b/os_dep/linux/ioctl_cfg80211.c     2020-12-29 01:38:41.898244478 +0200
@@ -7465,10 +7465,10 @@
 }

-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0))
-static void cfg80211_rtw_update_mgmt_frame_register(struct wiphy *wiphy,
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0)) || defined(BUILD_OPENWRT)
+static void cfg80211_rtw_update_mgmt_frame_registrations(struct wiphy *wiphy,
                                              struct wireless_dev *wdev,
                                              struct mgmt_frame_regs *upd)
 #else
 static void cfg80211_rtw_mgmt_frame_register(struct wiphy *wiphy,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 6, 0))
@@ -9876,8 +9879,8 @@

 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 37)) || defined(COMPAT_KERNEL_RELEASE)
        .mgmt_tx = cfg80211_rtw_mgmt_tx,
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0))
-       .update_mgmt_frame_registrations = cfg80211_rtw_update_mgmt_frame_register,
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || defined(BUILD_OPENWRT)
+       .update_mgmt_frame_registrations = cfg80211_rtw_update_mgmt_frame_registrations,
 #else
        .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
 #endif

```
and the Makefile, define Build/Compile modified like previous post

also you can directly checkout from git like

```
PKG_NAME:=rtl8812au-ac
PKG_RELEASE:=1
PKG_SOURCE_DATE:=2020-12-27

PKG_LICENSE:=GPLv2
PKG_LICENSE_FILES:=

PKG_SOURCE:=$(PKG_NAME)-$(PKG_SOURCE_DATE).tar.gz
PKG_SOURCE_VERSION:=de00413741782e7df2dd537e12d9fe03f72fd02b
PKG_SOURCE_URL:=https://github.com/aircrack-ng/rtl8812au.git
PKG_SOURCE_PROTO:=git

PKG_MIRROR_HASH:=7ecafc2a4678526b0b9093b818159d827a04d8202c27090cd5acca8708d2e8da

```

but for me the device is not detected on 5.7.0 (only with 5.6.4.2 branch, needs first line patch to be modified)

`[  524.465849] usbcore: registered new interface driver rtl8812au`

Bus 001 Device 004: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]

on 5.6.4.2
[   13.987042] usb 1-1.3: 88XXau e8:4e:06:xx:xx:xx hw_info[107]
[   13.996611] usbcore: registered new interface driver rtl88XXau

daemon.notice hostapd: wlan0: interface state UNINITIALIZED->ENABLED
daemon.notice hostapd: wlan0: AP-ENABLED

but no ip without encryption or can't connect using wpa
```
daemon.info hostapd: wlan0: STA 42:0f:ff:2a:6d:14 IEEE 802.11: associated
daemon.info hostapd: wlan0: STA 42:0f:ff:2a:6d:14 IEEE 802.11: deauthenticated due to local deauth request
daemon.info hostapd: wlan0: STA 42:0f:ff:2a:6d:14 IEEE 802.11: disassociated

```