aircrack-ng/rtl8812au #801: Error compiling on kernel 5.10

henres, 2020-12-29:
Hello,

tried this morning to compile and had warnings and errors.

```bash
$ sudo make                                               
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.1-162.current/build M=/home/henres/Apps/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.1-162.current'
  CC [M]  /home/henres/Apps/rtl8812au/os_dep/osdep_service.o
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c: In function ‘isFileReadable’:
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2201:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2201 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2201:11: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2203:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2203 |   set_fs(KERNEL_DS);
      |   ^~~~~~
      |   sget_fc
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2203:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2203 |   set_fs(KERNEL_DS);
      |          ^~~~~~~~~
      |          KERNFS_NS
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2203:10: note: each undeclared identifier is reported only once for each function it appears in
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2243:12: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
 2243 |    oldfs = get_fs();
      |            ^~~~~~
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2245:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2245 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c: In function ‘storeToFile’:
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2282:12: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
 2282 |    oldfs = get_fs();
      |            ^~~~~~
/home/henres/Apps/rtl8812au/os_dep/osdep_service.c:2284:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2284 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /home/henres/Apps/rtl8812au/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1805: /home/henres/Apps/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.1-162.current'
make: *** [Makefile:2246: modules] Error 2
```

My configuration is:

Solus
Kernel 5.10.1

CGarces, 2020-12-29:
Hi.

The issue was fixed on 4573749621508b6799b84e92d6b08505fc5a2a7e
Please use de default branch and try again with an updated git branch.

henres, 2020-12-30:
Everything working ! Thanks.

I close

closed 2020-12-30 by henres