aircrack-ng/rtl8812au #802: make dkms_install does not work (Error 1)

Aholicknight, 2020-12-30:
I am running the latest version of Kali Linux and trying to install the driver. I enter the command `sudo make dkms_install` and it shows this as the output.
```
mkdir -p /usr/src/8812au-5.6.4.2_35491.20191025
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025

Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source ->
                 /usr/src/8812au-5.6.4.2_35491.20191025

DKMS: add completed.
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Error! echo
Your kernel headers for kernel 5.9.0-kali1-amd64 cannot be found at
/lib/modules/5.9.0-kali1-amd64/build or /lib/modules/5.9.0-kali1-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:2306: dkms_install] Error 1
```

How can I fix this? The Kali Version I am currently using is 2020.4

Thank you!

giangiluca, 2020-12-30:
The Same problem.

giangiluca, 2020-12-30:
solved with this installation:
https://online-it.nu/how-to-install-alfa-awus1900-kali-linux-2018-4/

Aholicknight, 2020-12-31:
@giangiluca This works thank you.

Aholicknight, 2021-01-01:
I will keep this issue open as this problem still remains with the repo.

danieletorelli, 2021-01-04:
Happened to me on RPI 4 but for a different reason: it seems like the changes introduced by https://github.com/aircrack-ng/rtl8812au/pull/769 are resulting in a build failure `gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together`.

Full build log if you are interested:
```
DKMS make.log for 8812au-5.6.4.2_35491.20191025 for kernel 5.4.83-5-ARCH (armv7l)
Mon Jan  4 03:38:08 PM CET 2021
make[1]: Entering directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.4.83-5-ARCH/build M=/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build  modules
make[2]: Entering directory '/usr/lib/modules/5.4.83-5-ARCH/build'
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[3]: *** [scripts/Makefile.build:266: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o] Error 1
make[3]: *** Waiting for unfinished jobs....
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[3]: *** [scripts/Makefile.build:266: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o] Error 1
make[3]: *** [scripts/Makefile.build:266: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o] Error 1
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[3]: *** [scripts/Makefile.build:266: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o] Error 1
make[2]: *** [Makefile:1732: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build] Error 2
make[2]: Leaving directory '/usr/lib/modules/5.4.83-5-ARCH/build'
make[1]: *** [Makefile:2252: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
```

CGarces, 2021-01-13:
@danieletorelli I can confirm that the makefile if broken due #769 

arm kernels are build with  -mfloat-abi=soft 

https://github.com/torvalds/linux/blob/master/arch/arm/Makefile#L128

So -mfloat-abi=hard not works on arm builds with kernels compiled with arm 32b (works fine on arm64)

Will work on that on the weekend.

ping @pprindeville

pprindeville, 2021-01-13:
@CGarces What are you suggesting?

What would probably be cleanest is something like:

```
else ifeq ($(ARCH), arm)
# Raspbian kernel is with soft-float.
# 'softfp' allows FP instructions, but no FP on function call interfaces
ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
EXTRA_CFLAGS += -mfloat-abi=softfp
else ifeq ($(ARCH), arm64)
EXTRA_CFLAGS += -mfloat-abi=hard
else
$(error ARCH: need architecture explicitly set)
endif
endif
```

CGarces, 2021-01-14:
Hi @pprindeville 
Please review #814 and continue the discussion on that PR.
I open to any ideas my knowledge is limited on this topic.

danieletorelli, 2021-01-20:
Thanks for your fix @CGarces. I can confirm that now works as expected.

deadpool6t9, 2021-08-17:
> solved with this installation:
> https://online-it.nu/how-to-install-alfa-awus1900-kali-linux-2018-4/

Thank you so much, I have been trying for days, this finally worked

myshuker, 2024-07-05:
> solved with this installation: https://online-it.nu/how-to-install-alfa-awus1900-kali-linux-2018-4/

i have the same problem but this link dosent work anymore

Aholicknight, 2024-07-07:
> > solved with this installation: https://online-it.nu/how-to-install-alfa-awus1900-kali-linux-2018-4/
> 
> i have the same problem but this link dosent work anymore

@myshuker https://web.archive.org/web/20220412181012/https://online-it.nu/how-to-install-alfa-awus1900-kali-linux-2018-4/