aircrack-ng/rtl8812au #803: dkms fails with error code 10 on Kernel 5.10.3 Arch

cedricbambooza, 2020-12-31:
can't compile kernel module through dkms on new kernel on arch system

[make.log](https://github.com/aircrack-ng/rtl8812au/files/5757327/make.log)

cedricbambooza, 2020-12-31:
Output while running pacman -U looks like this: 

[neo@bladerunner rtl88xxau-aircrack-dkms-git]$ sudo pacman -U rtl88xxau-aircrack-dkms-git-r1123.e9fbf5c-1-x86_64.pkg.tar.zst 
[sudo] password for neo: 
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) rtl88xxau-aircrack-dkms-git-r1123.e9fbf5c-1

Total Installed Size:  50.86 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                                                            [###############################################################] 100%
(1/1) checking package integrity                                                                          [###############################################################] 100%
(1/1) loading package files                                                                               [###############################################################] 100%
(1/1) checking for file conflicts                                                                         [###############################################################] 100%
(1/1) checking available disk space                                                                       [###############################################################] 100%
:: Processing package changes...
(1/1) installing rtl88xxau-aircrack-dkms-git                                                              [###############################################################] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Install DKMS modules
==> dkms install --no-depmod -m rtl88xxau -v r1123.e9fbf5c -k 5.10.3-arch1-1
Error! Bad return status for module build on kernel: 5.10.3-arch1-1 (x86_64)
Consult /var/lib/dkms/rtl88xxau/r1123.e9fbf5c/build/make.log for more information.
==> Warning, `dkms install --no-depmod -m rtl88xxau -v r1123.e9fbf5c -k 5.10.3-arch1-1' returned 10
==> depmod 5.10.3-arch1-1

cedricbambooza, 2020-12-31:
tried the version directly from the github repo, that works well even though I haven't installed it through dkms.
So maybe the version from the AUR needs updating ^^

iyanmv, 2020-12-31:
Please close this issue. @cedricbambooza is not using the latest commit and AUR package is fine.

cedricbambooza, 2020-12-31:
Sorry my bet then ^^

closed 2020-12-31 by cedricbambooza