aircrack-ng/rtl8812au #804: Stacktrace on armv7h (rpi3b)

solsticedhiver, 2021-01-05:
On archlinux-arm with kernel 5.4.83-3-ARCH, and with driver from commit r1137 aka 2686ea8, I alays got now this stacktrace:
```
Dec 21 04:38:15 daphnis kernel: usb 1-1.3: new high-speed USB device number 7 using dwc_otg
Dec 21 04:38:15 daphnis kernel: 88XXau: loading out-of-tree module taints kernel.
Dec 21 04:38:15 daphnis kernel: BUG: FP instruction issued in kernel mode with FP unit disabled
Dec 21 04:38:15 daphnis kernel: Internal error: Oops - undefined instruction: 0 [#1] PREEMPT SMP ARM
Dec 21 04:38:15 daphnis kernel: Modules linked in: 88XXau(O+) brcmfmac brcmutil evdev snd_usb_audio snd_hwdep cfg80211 snd_usbmidi_lib snd_rawmidi snd_seq_device bcm283>
Dec 21 04:38:15 daphnis kernel: CPU: 3 PID: 235 Comm: systemd-udevd Tainted: G         C O      5.4.83-3-ARCH #1
Dec 21 04:38:15 daphnis kernel: Hardware name: BCM2835
Dec 21 04:38:15 daphnis kernel: PC is at usbctrl_vendorreq+0x4/0x240 [88XXau]
Dec 21 04:38:15 daphnis kernel: LR is at usb_read32+0x40/0x68 [88XXau]
Dec 21 04:38:15 daphnis kernel: random: crng init done
Dec 21 04:38:15 daphnis kernel: pc : [<7f4b8f98>]    lr : [<7f4f5ea4>]    psr: 20070013
Dec 21 04:38:15 daphnis kernel: random: 7 urandom warning(s) missed due to ratelimiting
Dec 21 04:38:15 daphnis kernel: sp : b1eb9cec  ip : 00000004  fp : 00000004
Dec 21 04:38:15 daphnis kernel: r10: 00000009  r9 : b13ce878  r8 : b8c47000
Dec 21 04:38:15 daphnis kernel: r7 : b21efa00  r6 : b2b32000  r5 : b8c44000  r4 : b8ccd000
Dec 21 04:38:15 daphnis kernel: r3 : 00000000  r2 : 000000f0  r1 : 00000005  r0 : b8c44f9c
Dec 21 04:38:15 daphnis kernel: Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Dec 21 04:38:15 daphnis kernel: Control: 10c5383d  Table: 31ebc06a  DAC: 00000055
Dec 21 04:38:15 daphnis kernel: Process systemd-udevd (pid: 235, stack limit = 0x(ptrval))
Dec 21 04:38:15 daphnis kernel: Stack: (0xb1eb9cec to 0xb1eba000)
Dec 21 04:38:16 daphnis kernel: 9ce0:                            b8ccd000 b8c44000 b2b32000 b21efa00 b8c47000
Dec 21 04:38:16 daphnis kernel: 9d00: b13ce878 00000009 00000004 7f4f5ea4 b1eb9d20 00000004 00000001 0000eb10
Dec 21 04:38:16 daphnis kernel: 9d20: 00000000 81204f88 b8ccd000 7f4fdd20 b8c44000 b8c47000 00000484 b8c44000
Dec 21 04:38:16 daphnis kernel: 9d40: 00000001 7f4d93b4 b8c44000 7f4b855c b2b32000 b21efa00 b13ce800 7f553604
Dec 21 04:38:16 daphnis kernel: 9d60: b2b32c44 7f4b87ec 011ba6e8 81204f88 00000000 b21efa20 b13ce878 7f55321c
Dec 21 04:38:16 daphnis kernel: 9d80: b21efa00 00000001 b13ce800 7f553604 ffffffed 80951dc4 80951cfc b21efa20
Dec 21 04:38:16 daphnis kernel: 9da0: 81387754 00000000 81387758 00000000 7f55321c 0000001e 011ba6e8 808c8b00
Dec 21 04:38:16 daphnis kernel: 9dc0: 00000000 b21efa20 7f55321c b21efa64 7f55321c b1eb8000 76d3126c 00000080
Dec 21 04:38:16 daphnis kernel: 9de0: 011ba6e8 808c8f84 7f553214 b3160158 b21efa20 00000000 b21efa64 7f55321c
Dec 21 04:38:16 daphnis kernel: 9e00: b1eb8000 808c9320 00000000 7f55321c b21efa20 812995f8 b1eb8000 808c93a0
Dec 21 04:38:16 daphnis kernel: 9e20: 00000000 7f55321c 808c9328 808c69c8 011ba6e8 b3160158 b0f6a3b4 81204f88
Dec 21 04:38:16 daphnis kernel: 9e40: 7f55321c b1eefb00 00000000 808c7d18 7f542150 80c34b84 7f55321c 7f55321c
Dec 21 04:38:16 daphnis kernel: 9e60: 00000000 7f55321c 812995f8 808c99dc 7f5531e4 7f542148 7f55321c 80950b74
Dec 21 04:38:16 daphnis kernel: 9e80: 7f244000 00000000 7f5531e4 b92abee8 b1eb8000 7f244048 7f244000 751cbef0
Dec 21 04:38:16 daphnis kernel: 9ea0: 00000000 80102e44 00030012 0000e623 00000000 00000000 00000000 00000000
Dec 21 04:38:16 daphnis kernel: 9ec0: 00000000 00000000 6e72656b 00006c65 00000000 00000000 00000000 00000000
Dec 21 04:38:16 daphnis kernel: 9ee0: b3001e00 b1a33040 00000000 7f5b5600 751cbef0 81204f88 b92abee8 7f5b5600
Dec 21 04:38:16 daphnis kernel: 9f00: 751cbef0 b1a33000 b92abee8 801c6424 751cbef0 00000000 b92abee8 0000fee8
Dec 21 04:38:16 daphnis kernel: 9f20: 751cbef0 00000000 b92abee8 801c8f00 b9189a91 b91edd40 b908c000 0021fee8
Dec 21 04:38:16 daphnis kernel: 9f40: b92ab880 b92ab6e8 b92041a8 0016d000 0017af70 00023f60 0018e411 00000000
Dec 21 04:38:16 daphnis kernel: 9f60: 00000000 00000000 00023f50 00000026 00000027 0000001e 00000000 0000000f
Dec 21 04:38:16 daphnis kernel: 9f80: 00000000 81204f88 e3784300 011df580 74fac008 00020000 00000080 80101204
Dec 21 04:38:16 daphnis kernel: 9fa0: b1eb8000 801011d4 011df580 74fac008 74fac008 0021fee8 76d3126c e3784300
Dec 21 04:38:16 daphnis kernel: 9fc0: 011df580 74fac008 00020000 00000080 011e1d00 00000000 76f704ba 011ba6e8
Dec 21 04:38:16 daphnis kernel: 9fe0: 76d41e00 7ec6d630 76d289a0 76c13230 60070010 74fac008 00000000 00000000
Dec 21 04:38:16 daphnis kernel: unwind: Unhandled instruction c9
Dec 21 04:38:16 daphnis kernel: Code: ebffed54 eaffff6d 00007fe8 e92d4ff0 (ed2d8b02) 
Dec 21 04:38:16 daphnis kernel: ---[ end trace 3e7dbdf3a70aa888 ]---

```

yasij, 2021-01-11:
This also happened to me on a rpi2B on Ubuntu 20.10 with 5.8.0-1011-raspi. I reverted 3325543 and it worked.

solsticedhiver, 2021-01-11:
3325543 is assuming all arm rpi are soft-float. this is wrong.

 ```
$ readelf -a /lib/libc.so.6  | grep VFP
  Tag_FP_arch: VFPv3-D16
  Tag_ABI_VFP_args: VFP registers
```
The last line seems to imply I have hard-float arch on archlinux-arm (armv7h aka rpi3b)

tomty89, 2021-01-11:
@solsticedhiver Yet the driver cannot be compiled with `-mfloat-abi=hard`: https://github.com/aircrack-ng/rtl8812au/issues/802#issuecomment-754014418

Builds and works on ArchLinuxARM (RPI 4B) without specifying `-mfloat-abi=` at all.