aircrack-ng/rtl8812au #808: Frequent disconnects on kernel 5.10.2 (with logs)

GodTamIt, 2021-01-08:
Getting some pretty frequent disconnects with the TP-Link Archer T4Uv3. This (visually) manifested itself by repeatedly trying to reconnect to Wi-Fi (NetworkManager) and the Gnome NM front-end would repeatedly prompt me for the password.

Here are the logs/backtrace that seem relevant:

```
[ 4868.462233] WARNING: CPU: 1 PID: 180 at net/wireless/sme.c:757 __cfg80211_connect_result+0x3db/0x420 [cfg80211]
[ 4868.462235] Modules linked in: rndis_host cdc_ether usbnet mii nls_utf8 isofs sr_mod cdrom uas usb_storage uhid rfcomm cmac algif_hash algif_skcipher af_alg bnep btusb btrtl btbcm btintel bluetooth ecdh_generic ecc crc16 hid_logitech_h
idpp mousedev joydev hid_logitech_dj snd_usb_audio snd_usbmidi_lib snd_rawmidi snd_seq_device mc usbhid 88XXau(OE) 88x2bu(OE) intel_rapl_msr intel_rapl_common cfg80211 rfkill snd_hda_codec_realtek nct7802 lm78 snd_hda_codec_generic hwmon_
vid f71882fg ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg x86_pkg_temp_thermal soundwire_intel intel_powerclamp soundwire_generic_allocation coretemp soundwire_cadence kvm_intel snd_hda_codec snd_hda_core vfat snd_hwdep
 fat kvm soundwire_bus irqbypass snd_soc_core crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel iTCO_wdt crypto_simd intel_pmc_bxt snd_compress cryptd iTCO_vendor_support mxm_wmi ac97_bus glue_helper snd_pcm_dmaengine rapl int
el_cstate snd_pcm uinput nvidia_drm(POE) i2c_i801
[ 4868.462330]  snd_timer nvidia_modeset(POE) snd mei_me intel_uncore e1000e i2c_smbus mei lpc_ich soundcore drm agpgart wmi mac_hid nvidia(POE) vboxnetflt(OE) vboxnetadp(OE) vboxdrv(OE) crypto_user fuse ip_tables x_tables btrfs blake2b_g
eneric libcrc32c crc32c_generic xor raid6_pq xhci_pci crc32c_intel
[ 4868.462369] CPU: 1 PID: 180 Comm: kworker/u24:8 Tainted: P        W  OE     5.10.2-2-MANJARO #1
[ 4868.462371] Hardware name: EVGA INTERNATIONAL CO.,LTD Default string/131-HE-E995, BIOS 2.08 06/28/2019
[ 4868.462395] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[ 4868.462431] RIP: 0010:__cfg80211_connect_result+0x3db/0x420 [cfg80211]
[ 4868.462435] Code: 8b 40 30 48 85 c0 0f 84 58 fd ff ff f0 ff 40 e8 e9 4f fd ff ff 0f 0b 48 8b 76 10 48 8b 3b e8 9c 07 fd ff e9 2a fe ff ff 0f 0b <0f> 0b e9 21 fe ff ff e8 79 4c 3d ca e9 17 fe ff ff 0f 0b e9 41 fd
[ 4868.462438] RSP: 0018:ffffabeb405ffdf0 EFLAGS: 00010246
[ 4868.462441] RAX: 0000000000000000 RBX: ffff8d7664209800 RCX: 0000000000000003
[ 4868.462443] RDX: 0000000000000002 RSI: 00000000fffffe00 RDI: ffffffffc270f0db
[ 4868.462445] RBP: ffff8d77a6907e18 R08: ffff8d77a2f3d0c0 R09: fffffffeffeb399d
[ 4868.462447] R10: ffff8d77a6907e88 R11: ffff8d7664209870 R12: ffff8d77a2f3f000
[ 4868.462449] R13: ffffabeb405ffdf0 R14: dead000000000122 R15: dead000000000100
[ 4868.462452] FS:  0000000000000000(0000) GS:ffff8d7d1fa40000(0000) knlGS:0000000000000000
[ 4868.462454] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 4868.462456] CR2: 00007fccf45f6000 CR3: 0000000176f7a004 CR4: 00000000001706e0
[ 4868.462459] Call Trace:
[ 4868.462491]  cfg80211_process_wdev_events+0x139/0x1a0 [cfg80211]
[ 4868.462517]  cfg80211_process_rdev_events+0x32/0x70 [cfg80211]
[ 4868.462540]  cfg80211_event_work+0x1a/0x20 [cfg80211]
[ 4868.462547]  process_one_work+0x1d6/0x3a0
[ 4868.462555]  worker_thread+0x4d/0x3d0
[ 4868.462560]  ? rescuer_thread+0x410/0x410
[ 4868.462563]  kthread+0x133/0x150
[ 4868.462567]  ? __kthread_bind_mask+0x60/0x60
[ 4868.462572]  ret_from_fork+0x22/0x30
[ 4868.462577] ---[ end trace 97e44ebc3833c412 ]---
```

```
[ 4873.188226] WARNING: CPU: 0 PID: 9545 at /var/lib/dkms/88x2bu/1.1/build/core/rtw_mlme.c:2669 rtw_sta_mstatus_disc_rpt+0x97/0xbc [88x2bu]
[ 4873.188227] Modules linked in: rndis_host cdc_ether usbnet mii nls_utf8 isofs sr_mod cdrom uas usb_storage uhid rfcomm cmac algif_hash algif_skcipher af_alg bnep btusb btrtl btbcm btintel bluetooth ecdh_generic ecc crc16 hid_logitech_hidpp mousedev joydev hid_logitech_dj snd_usb_audio snd_usbmidi_lib snd_rawmidi snd_seq_device mc usbhid 88XXau(OE) 88x2bu(OE) intel_rapl_msr intel_rapl_common cfg80211 rfkill snd_hda_codec_realtek nct7802 lm78 snd_hda_codec_generic hwmon_vid f71882fg ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg x86_pkg_temp_thermal soundwire_intel intel_powerclamp soundwire_generic_allocation coretemp soundwire_cadence kvm_intel snd_hda_codec snd_hda_core vfat snd_hwdep fat kvm soundwire_bus irqbypass snd_soc_core crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel iTCO_wdt crypto_simd intel_pmc_bxt snd_compress cryptd iTCO_vendor_support mxm_wmi ac97_bus glue_helper snd_pcm_dmaengine rapl intel_cstate snd_pcm uinput nvidia_drm(POE) i2c_i801
[ 4873.188269]  snd_timer nvidia_modeset(POE) snd mei_me intel_uncore e1000e i2c_smbus mei lpc_ich soundcore drm agpgart wmi mac_hid nvidia(POE) vboxnetflt(OE) vboxnetadp(OE) vboxdrv(OE) crypto_user fuse ip_tables x_tables btrfs blake2b_generic libcrc32c crc32c_generic xor raid6_pq xhci_pci crc32c_intel
[ 4873.188286] CPU: 0 PID: 9545 Comm: RTW_CMD_THREAD Tainted: P        W  OE     5.10.2-2-MANJARO #1
[ 4873.188287] Hardware name: EVGA INTERNATIONAL CO.,LTD Default string/131-HE-E995, BIOS 2.08 06/28/2019
[ 4873.188308] RIP: 0010:rtw_sta_mstatus_disc_rpt+0x97/0xbc [88x2bu]
[ 4873.188310] Code: c7 c1 74 96 93 c2 4c 0f 45 c1 89 c1 48 c7 c6 50 45 92 c2 48 c7 c7 75 96 93 c2 e8 4a fe c2 ca eb 94 83 3d f9 7a 24 00 00 75 05 <0f> 0b 5b 5d c3 40 0f b6 ce 48 8b 97 28 46 00 00 48 c7 c6 50 45 92
[ 4873.188311] RSP: 0018:ffffabeb4775fe40 EFLAGS: 00010246
[ 4873.188312] RAX: ffff8d77a2f3b000 RBX: 00000000000000ff RCX: 0000000000000000
[ 4873.188313] RDX: 0000000000000010 RSI: 00000000000000ff RDI: ffffabeb404f3000
[ 4873.188313] RBP: ffffabeb404f3000 R08: 0000000000000000 R09: 0000000000000000
[ 4873.188314] R10: ffffabeb47e17a80 R11: 0000000000000000 R12: ffffabeb404f48d0
[ 4873.188315] R13: ffffabeb404f4118 R14: ffff8d77a5591ec0 R15: ffffffffc2824902
[ 4873.188316] FS:  0000000000000000(0000) GS:ffff8d7d1fa00000(0000) knlGS:0000000000000000
[ 4873.188317] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 4873.188318] CR2: 00007f9d19e3d000 CR3: 000000017d2ac005 CR4: 00000000001706f0
[ 4873.188318] Call Trace:
[ 4873.188342]  rtw_stadel_event_callback+0x31/0x32d [88x2bu]
[ 4873.188364]  ? set_tx_beacon_cmd+0xc4/0xc4 [88x2bu]
[ 4873.188384]  mlme_evt_hdl+0x67/0x7e [88x2bu]
[ 4873.188401]  rtw_cmd_thread+0x41e/0x5ce [88x2bu]
[ 4873.188417]  ? rtw_stop_cmd_thread+0x39/0x39 [88x2bu]
[ 4873.188421]  kthread+0x133/0x150
[ 4873.188422]  ? __kthread_bind_mask+0x60/0x60
[ 4873.188426]  ret_from_fork+0x22/0x30
[ 4873.188428] ---[ end trace 97e44ebc3833c413 ]---
```

CGarces, 2021-01-12:
Hi @GodTamIt 
Probably is not related but, can you try again with the new code?, after #810

GodTamIt, 2021-01-13:
@CGarces it's actually been stable for the last 14+ hours. I'll update here if there's anything that pops up.

GodTamIt, 2021-01-14:
Update: seems we're still getting the issue even with the changes.

CGarces, 2021-01-14:
> Update: seems we're still getting the issue even with the changes.

Was a long shot, thanks for testing.