aircrack-ng/rtl8812au #809: compability kernel-5.10.x and rtl8812au

Elleni57, 2021-01-09:
I am using this driver for my [usb-wifi-stick](https://www.asus.com/ch-de/Networking/USB-AC68/) and it works fine on >=kernel-5.9.8 (gentoo) but stopped working on 5.10.X. Is there a newer version or do I need another module for recent kernels >=5.10?

CGarces, 2021-01-12:
Hi @GodTamIt 
Probably is not related but, can you try again with the new code?, after #810

Elleni57, 2021-01-31:
Tried with kernel-5.10.12 (gentoo-sources) - still the same. Anyone an idea how to fix this? My last working kernel for this module ist 5.9.8. If I should provide more information please tell me what I could help and I'll try my best to provide it.

hkroger, 2021-03-06:
I guess it depends on the device but if you used different branch to get it compiled might be that this PR #839 actually might fix it. Different branches detect different devices because the USB identifiers are not listed in all branches.

cedricbambooza, 2021-05-10:
Maybe some solution resulting from here might help you: #855 
Or if your issue is solved consider closing the issue :)