aircrack-ng/rtl8812au #817: Please add frequecy from 5340 to 5460

Zanzaux, 2021-01-17:
is it possible to add frequency  from  5340-5460?
kali 2020.4

    Frequencies:
                        * 5180 MHz [36] (30.0 dBm)
                        * 5200 MHz [40] (30.0 dBm)
                        * 5220 MHz [44] (30.0 dBm)
                        * 5240 MHz [48] (30.0 dBm)
                        * 5260 MHz [52] (30.0 dBm)
                        * 5280 MHz [56] (30.0 dBm)
                        * 5300 MHz [60] (30.0 dBm)
                        * 5320 MHz [64] (30.0 dBm)
                        * 5500 MHz [100] (30.0 dBm)
                        * 5520 MHz [104] (30.0 dBm)
                        * 5540 MHz [108] (30.0 dBm)
                        * 5560 MHz [112] (30.0 dBm)
                        * 5580 MHz [116] (30.0 dBm)
                        * 5600 MHz [120] (30.0 dBm)
                        * 5620 MHz [124] (30.0 dBm)
                        * 5640 MHz [128] (30.0 dBm)
                        * 5660 MHz [132] (30.0 dBm)
                        * 5680 MHz [136] (30.0 dBm)
                        * 5700 MHz [140] (30.0 dBm)
                        * 5720 MHz [144] (30.0 dBm)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
                        * 5825 MHz [165] (30.0 dBm)
                        * 5845 MHz [169] (30.0 dBm)
                        * 5865 MHz [173] (30.0 dBm)
                        * 5885 MHz [177] (30.0 dBm)

astsam, 2021-01-25:
Try this patch for v5.6.4.2.
[0001-Add-missing-5GHz-channels.txt](https://github.com/aircrack-ng/rtl8812au/files/5866557/0001-Add-missing-5GHz-channels.txt)

Zanzaux, 2021-01-25:
> Try this patch for v5.6.4.2.
> [0001-Add-missing-5GHz-channels.txt](https://github.com/aircrack-ng/rtl8812au/files/5866557/0001-Add-missing-5GHz-channels.txt)

Hello Thanks for taking an interest in the speech, where and how do I apply this file?

astsam, 2021-01-25:
1. Download the source code of the branch v5.6.4.2 (see section [Notes](https://github.com/aircrack-ng/rtl8812au#notes)).
2. Apply the patch
`patch -p1 < 0001-Add-missing-5GHz-channels.txt`
3. [Build and install](https://github.com/aircrack-ng/rtl8812au#make) the driver or use [DKMS](https://github.com/aircrack-ng/rtl8812au#dkms).

Zanzaux, 2021-01-25:
> 1. Download the source code of the branch v5.6.4.2 (see section [Notes](https://github.com/aircrack-ng/rtl8812au#notes)).
> 2. Apply the patch
>    `patch -p1 < 0001-Add-missing-5GHz-channels.txt`
> 3. [...]

THANKS ENDLESS !!! finally everything works and I don't get the message "kernel reports: Channel is disabled" when I try to set the channel
Thanks again I hope the aircrack-ng team adds your default patch.
Thanks.

Zanzaux, 2021-01-26:
> 1. Download the source code of the branch v5.6.4.2 (see section [Notes](https://github.com/aircrack-ng/rtl8812au#notes)).
> 2. Apply the patch
>    `patch -p1 < 0001-Add-missing-5GHz-channels.txt`
> 3. [...]

Can other frequencies be enabled? like below 5180 and above 5885?

astsam, 2021-01-27:
Technically this is possible, but the driver tunes some undocumented registers depending on RF band. Due to the lack of the documentation this modification may cause suboptimal performance in the extended frequency range.

Zanzaux, 2021-01-27:
Hello astsam  If you can enable  them I would be curious to carry out some tests perhaps adding from 4900 to 5175,

astsam, 2021-01-27:
I can try to enable 5075 to 5175 frequency range.

Zanzaux, 2021-01-27:
> I can try to enable 5075 to 5175 frequency range.

Okay thanks, even if I get lower performance I don't care, I have to test some antennas and I'm on non-standard frequencies (4900,5010,5060,5340,5920,5940,6000)

Thanks again

astsam, 2021-01-27:
[0002-Add-low-frequency-5GHz-band.txt](https://github.com/aircrack-ng/rtl8812au/files/5882288/0002-Add-low-frequency-5GHz-band.txt)
It will be great if you check the real RF output with a spectrum analyzer.

Zanzaux, 2021-01-28:
Thanks astsam :), let's see if I can do some tests with the spectrum analyzer this weekend. With airodump-ng an access point I was testing on Ch 5320 at a distance of about 1, Km had a signal of -30, -33 db, now this Ap is on Ch 5100 and I detect it at -42db.
I noticed that when I set the channel under 5180 by typing iwconfig
I detect channel "0", which does not happen with the other channels where the set channel is shown like 5340,5200,5480 etc ...

airodump-ng wlan1 -C 5100
iwconfig
wlan1 unassociated ESSID: "" Nickname: "<WIFI @ REALTEK>"
          Mode: Monitor **Channel = 0** Access Point: Not-Associated
          Sensitivity: 0/0
          Retry: off RTS thr: off Fragment thr: off
          Encryption key: off
          Power Management: off
          Link Quality = 0/100 Signal level = 0 dBm Noise level = 0 dBm
          Rx invalid nwid: 0 Rx invalid crypt: 0 Rx invalid frag: 0
          Tx excessive retries: 0 Invalid misc: 0 Missed beacon: 0

astsam, 2021-01-28:
I used iw command to set and get channel. You are right, iwconfig has incorrect output.
```
$sudo iw wlx000f00371234 info
Interface wlx000f00371234
	ifindex 58
	wdev 0x3700000001
	addr 00:0f:00:37:12:34
	type monitor
	wiphy 55
	channel 15 (5075 MHz), width: 20 MHz, center1: 5075 MHz
	txpower 20.00 dBm
```
```
$ sudo iwconfig wlx000f00371234
wlx000f00371234  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Channel=0  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
I supposed the iwconfig is deprecated and see no reason to fix this issue. Try to use the iw utility.

Zanzaux, 2021-01-30:
Thanks astsam, I made some tests through "airodump-ng wlan1 -C ch" here are the values ​​I get by varying the frequencies from the AP

-52 4920
-48 4950
-1 4970
-45 4990
-1 5010 (don't detected ap) Freq 5010] [Elapsed: 12 s] [fixed frequency wlan1: 2417  (important channel***)
-1 5060
-45 5075
-41 5090

-44 5100
-34 5130
-34 5160
-34 5180

-33 5320
-33 5380
-33 5460
-33 5500

-43 5700
-39 5785

-42 5925
-42 5950
-42 5990
-42 6000

here is also the reg domain file I used
        (2400 - 2494 @ 20), (N / A, 30), (N / A)
        (4920 - 4995 @ 80), (N / A, 33), (N / A)
        (5000 - 5145 @ 80), (N / A, 33), (N / A)
        (5150 - 5465 @ 80), (N / A, 33), (N / A)
        (5470 - 5885 @ 160), (N / A, 33), (N / A)
        (5900 - 6100 @ 160), (N / A, 33), (N / A)
        (57000 - 66000 @ 2160), (N / A, 40), (N / A)

If you can add from 4920-5070 it would be the top !!!
Thank you so much for creating the patch driver and most of all Thanks for the amazing help and availability.

astsam2, 2021-01-30:
I am happy that my patches help you.
Zanzaux, will you clarify the meaning of the first values
```
-45 5075
...
```

astsam2, 2021-01-30:
> If you can add from 4920-5070 it would be the top

It is not clear for me how to support this range. The drive internally uses channel number instead of frequency.  Channels from 1 to 14 is reserved for 2,4GHz band. Moreover, due to this formula frequency less than 5000 MHz corresponds to negative channel value.
```
Freq = (Chan * 15) + 5000
```
Therefore, the lowest frequency that could be added is 5075 MHz (corresponding to channel 15).

Zanzaux, 2021-01-30:
> Sono felice che i miei cerotti ti aiutino.
> Zanzaux, chiarirai il significato dei primi valori
> 
> ```
> -45 5075
> ...
> ```
Se ti riferisci ai primi valori come  4920,4950,4970,4990,5010,5060 li avevo menzionati se potevano essere aggiunti in una delle tue incredibili patch :)  :)  :)

Zanzaux, 2021-01-30:
> Sono felice che i miei cerotti ti aiutino.
> Zanzaux, chiarirai il significato dei primi valori
> 
> ```
> -45 5075
> ...
> ```

If you refer to the first values ​​like 4920,4950,4970,4990,5010,5060 I had mentioned them if content be added in one of your amazing patches :) :) :)

Zanzaux, 2021-01-30:
> It is not clear for me how to support this range. The drive internally uses channel number instead of frequency. Channels from 1 to 14 is reserved for 2,4GHz band. Moreover, due to this formula [...]

I understand, I noticed that through airodump-ng by setting the frequency 5010 it is shown to me as channel "2" which as you say belongs to 2.4 instead the frequency 4970 is shown as channel "15" do not worry I'm sure that if you will find the right documentation you will have no problems enabling them :) you have really done a lot for me congratulations and thanks !!!

astsam2, 2021-01-31:
`If you refer to the first values ​​like 4920,4950,4970,4990,5010,5060 I had mentioned them if content be added in one of your amazing patches :)`
Sorry Zanzauz, English is not may native language. Will you clarify the meaning of -45 value in my example. Is any a difference between a desired frequency and the real frequency? I am afraid my patch is not ensure the exact equality the desired frequency and the real frequency in the band below channel 36. I cannot check this myself because I have not a proper measurement equipment.

Zanzaux, 2021-02-02:
Hi astsam I will do an analysis with the spectrum analyzer, since I have never created an access point via linux and network card, do you have any program to advise me on how to create it?

Zanzaux, 2021-02-02:
Hi astsam I will do an analysis with the spectrum analyzer

Zanzaux, 2021-02-04:
Unfortunately I am having problems in creating an access point via linux under channel 36, if I insert the channels from 36 more I can create an access point, I have tried both with the linux manager and with hostapd-dnsmasq.
Do you have any solution?

astsam2, 2021-02-05:
It looks like the hostapd has additional checks for channel validity.

astsam2, 2021-02-05:
To send frames you may use scapy and a script like this. It sends ProbeRequest frames via mon0 interface.
```
#!/usr/bin/env python3
from scapy.all import *
from scapy.layers.dot11 import *

def main():
    receiver_mac = 'ff:ff:ff:ff:ff:ff'
    # bssid = receiver_mac
    bssid = '03:bb:bb:bb:bb:bb'
    transmitter_mac = '00:22:12:34:56:78'
    ssid = '12345678'
    channel = chr(11)
    iface = 'mon0'
    frame_count = 1000

    frame = RadioTap() \
            / Dot11(type=0, subtype=4, addr1=receiver_mac, addr2=transmitter_mac, addr3=bssid) \
            / Dot11ProbeReq() \
            / Dot11Elt(ID='SSID', info=ssid) \
            / Dot11Elt(ID='Rates', info='\x82\x84\x8b\x96\x0c\x12\x18') \
            / Dot11Elt(ID='ESRates', info='\x30\x48\x60\x6c') \
            / Dot11Elt(ID='DSset', info=channel)
#    answer = srp1(frame, iface=iface)
#    wireshark(frame)

    # send frame_count frames
    # sendp(frame, iface=iface, inter=0.100, count=frame_count)

    # send frames indefinitely
    sendp(frame, iface=iface, inter=0.100, loop=1)

# Press the green button in the gutter to run the script.
if __name__ == '__main__':
    main()
```

Zanzaux, 2021-02-06:
> The driver has some checks in ap logic that I missed. Will you try this patch?
> [0001-Fix-error-creating-an-access-point-below-channel-36.txt](https://github.com/aircrack-ng/rtl8812au/files/5935155/0001-Fix-error-creating-an-access-point-below-channel-36.txt)

installing the patch and then the driver via dkms when restarting the network card was no longer present, i just downloaded your driver with the patch applied and now the access point under channel 36 works,
can you correct the "NOTES" section download link on your repository  ? https://github.com/astsam2/rtl8812au

Zanzaux, 2021-02-06:
Here is the measurement:

CH 48 5240 -52
CH 46 Disable by hostpad
CH 44 5220 -51
CH 42 Disable by hostpad
CH 40 5200 -58
CH 38 Disable by hostpad

UNDER CH 36

CH 36 5180 -54
CH 32 5160 -56
CH 28 5140 -60
CH 24 5120 -58
CH 20 5100 -62
CH 18 5090 -63
Ch 17 5085 -61
CH 16 5080 -61
CH 15 5075 -61

Zanzaux, 2021-02-16:
Hi astsam typing "iw list" I arrive at the frequency of 5885, but I noticed that I can actually test up to 6000 mhz, could you add the remaining 100 mhz?

kaselix, 2021-02-19:
> It is not clear for me how to support this range. The drive internally uses channel number instead of frequency. Channels from 1 to 14 is reserved for 2,4GHz band. Moreover, due to this formula [...]

Hello congratulations for the incredible work, finally a driver with a reputable frequency list. I don't know if it can be useful but by scanning the channel for example 2327 I can "start" a deauth on an access point on the frequency of 4920, I don't know if the attack actually comes completed ...

Here is a list of the possible frequencies I have tested
4920 2327 4925 2332 4960 2367 4970 2377 4975 2382 4980 2387

I hope you will be able to increase the frequency range further.
Thanks.

lnvlna, 2024-09-10:
I got the start with 5080, what about 5000, 5020, 5040, 5060, any ideas, even if I completely disable 2.4? if you look at the formula, I can't use 0 for 5000, that is, at least 1 = 5005, and then 4 = 5020, 8 = 5040, 12 = 5060