aircrack-ng/rtl8812au #819: Aircrack-ng and mdk4 deauthintication are not working in this version

khalidzaatari, 2021-01-20:
When I run the aireplay-ng deauth attack, the results are always 0 ACKs for the target client. Same goes when using mdk4 it is unable to disconnect clients. The old version of this driver used to work. When I updated to this version it seems there is a bug. Thanks for the support

astsam, 2021-01-24:
The v5.6.4.2 filtered out any injected management and control frames. Try this patch.
[0001-Allow-inject-management-and-control-frames.txt](https://github.com/aircrack-ng/rtl8812au/files/5862527/0001-Allow-inject-management-and-control-frames.txt)

khalidzaatari, 2021-01-28:
Can you please explain for me how can I add or install this patch provided?Thanks!

astsam, 2021-01-29:
1. Download the source code of the branch v5.6.4.2 (see section [Notes](https://github.com/aircrack-ng/rtl8812au#notes)).
2. Download the patch.
3. Apply the patch
`patch -p1 < 0001-Allow-inject-management-and-control-frames.txt`
4. [Build and install](https://github.com/aircrack-ng/rtl8812au#make) the driver or use [DKMS](https://github.com/aircrack-ng/rtl8812au#dkms).

khalidzaatari, 2021-01-29:
It worked like a charm. Thanks!

closed 2021-01-29 by khalidzaatari

snoopspy, 2021-02-01:
https://youtu.be/jq2-BU5C_PI

ExpertInLinux, 2021-02-01:
@snoopspy Does this work with the RTL8814AU too?

MTKFX, 2022-02-05:
I can't use the patch , it shows a message that says : "Reversed ( or previously applied) patch detected!" . Can someone help me ?

mkarne1, 2023-04-25:
@astsam I installed the [latest driver](https://github.com/aircrack-ng/rtl8812au) but can't deauthenticate any client from any Wifi. 

I installed the driver with the following commands: 

```
sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-`uname -r`
reboot
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl
make dkms_install  
```

Afterwards, I ran:

```
airmon-ng check kill
ip link set wlan0 down
iw dev wlan0 set type monitor
ip link set wlan0 up
airodump-ng --bssid D4:3D:F3:74:F3:91 --channel 5 --write /tmp/test wlan0
```

And in a new terminal:

```
aireplay-ng --deauth 1000 -a D4:3D:XX:XX:XX:91 -c 14:7D:XX:XX:XX:44 wlan0          
19:46:04  Waiting for beacon frame (BSSID: D4:3D:F3:74:F3:91) on channel 5
19:46:05  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [31|73 ACKs]
19:46:05  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [74|89 ACKs]
19:46:06  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [136|118 ACKs]
19:46:06  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 2|63 ACKs]
19:46:07  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 0|63 ACKs]
19:46:07  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 1|64 ACKs]
19:46:08  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [14|65 ACKs]
19:46:08  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [89|89 ACKs]
19:46:09  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [74|88 ACKs]
19:46:09  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [128|119 ACKs]
```

It just keeps sending deauth packets but the client isn't deauthenticating. 

And I tried `patch -p1 < 0001-Allow-inject-management-and-control-frames.txt` but I get `zsh: no such file or directory: 0001-Allow-inject-management-and-control-frames.txt`

Any help would be really appreciated!

Thanks!