aircrack-ng/rtl8812au #820: Error on build with kernel 5.10.0-1-amd64

matteodisabatino, 2021-01-23:
Hi,
I'm triyng to build with kernel 5.10.0-1-amd64 but I can't since I receive some errors. Here the whole stack:

```bash
DKMS make.log for rtl8812au-5.6.4.2 for kernel 5.10.0-1-amd64 (x86_64)
sab 23 gen 2021, 10:52:47, CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-1-amd64/build M=/var/lib/dkms/rtl8812au/5.6.4.2/build  modules
make[1]: ingresso nella directory «/usr/src/linux-headers-5.10.0-1-amd64»
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/mlme_linux.o
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c: In function ‘isFileReadable’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2201:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2201 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2201:11: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2203:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2203 |   set_fs(KERNEL_DS);
      |   ^~~~~~
      |   sget_fc
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2203:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2203 |   set_fs(KERNEL_DS);
      |          ^~~~~~~~~
      |          KERNFS_NS
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2203:10: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2243:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2243 |    oldfs = get_fs();
      |            ^~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2245:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2245 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c: In function ‘storeToFile’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2282:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2282 |    oldfs = get_fs();
      |            ^~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2284:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2284 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.10.0-1-common/scripts/Makefile.build:284: /var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.o] Errore 1
make[3]: *** Attesa per i processi non terminati....
make[2]: *** [/usr/src/linux-headers-5.10.0-1-common/Makefile:1817: /var/lib/dkms/rtl8812au/5.6.4.2/build] Errore 2
make[1]: *** [/usr/src/linux-headers-5.10.0-1-common/Makefile:185: __sub-make] Errore 2
make[1]: uscita dalla directory «/usr/src/linux-headers-5.10.0-1-amd64»
make: *** [Makefile:2246: modules] Errore 2
```

Does anyone know any fix?

Cheers.

CGarces, 2021-01-23:
That issue was fixed on #773
Please check that you are using the last version of the source code.

matteodisabatino, 2021-01-23:
Very strange... I did the git pull before the building. What am I missing?

matteodisabatino, 2021-01-24:
The problem solved itself. I uninstalled the driver for kernels, I re-compiled and installed the driver again.
I really don't why it worked, but... it worked.

closed 2021-01-24 by matteodisabatino