aircrack-ng/rtl8812au #824: Packet injection with awus 1900 not working using kernel 5.10.0-kali2-amd64

edgarc96, 2021-01-30:


Cleax21, 2021-01-30:
Having the same problem, Able to set the wifi adapter to monitor mode. but packet injection is something that isn't working.

Sitting on the same version as @edgarc96

edgarc96, 2021-01-30:
I have read in other forums and chats that Parrot Os it is compatible with awus1900 chipset so ill try it today.

Cleax21, 2021-01-31:
@edgarc96 did it work?

astsam2, 2021-01-31:
AFAIK, the AWUS 1900 adapter is based on the RTL8814AU chipset. For this chipset check out another [driver](https://github.com/aircrack-ng/rtl8814au).

edgarc96, 2021-02-06:
Yes, it worked, you have to download the drivers from rokland will redirect you to github page, after that you have to test the injection with aireplay-ng -9 (interface) ..... If you use aireplay-ng --deauth (interface) will not work or send any package

edgarc96, 2021-02-06:
You have to use -9 instead of --test , sorry ..