aircrack-ng/rtl8812au #831: Can't connect 5Ghz network

MarceloZapatta, 2021-02-15:
Hello!

I'm having an issue, I can't connect to 5Ghz network, only to 2Ghz. In the 5Ghz keep asking to put password or just show "configuring interface". Sometimes I was able to connect to the 5 but I really don't know how to debug the connection, I'm trying to connect via KDE Network Interface.

Sometimes I see more than one network of the same wireless, can it be related?

[image]

I'm on KDE Plasma, linux kernel 4.19

Can you guys help me?

Thanks

clrOne, 2021-02-17:
+1 . Having the same issue

4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux

MarceloZapatta, 2021-02-17:
I was able to connect by fixing the channel in the router settings, now it is on 153.
I think that was the problem. 

But I don't try again on this package, ended up using the: [](https://github.com/gnab/rtl8812au)

closed 2021-02-17 by MarceloZapatta