aircrack-ng/rtl8812au #836: Unable to build - ‘wdev’ undeclared

justas-, 2021-02-19:
Trying to build from branch v5.6.4.2 (b65dcf4105641716d16f3a6c96507fdd9c1862b4)
Get error:
/home/user/aircrack-ng/rtl8812au/os_dep/linux/os_intfs.c: In function ‘netdev_vir_if_close’:

> /home/user/aircrack-ng/rtl8812au/os_dep/linux/os_intfs.c:2820:2: error: ‘wdev’ undeclared (first use in this function); did you mean ‘cdev’?
>  2820 |  wdev->iftype = NL80211_IFTYPE_MONITOR;
>       |  ^~~~
>       |  cdev
> /home/user/aircrack-ng/rtl8812au/os_dep/linux/os_intfs.c:2820:2: note: each undeclared identifier is reported only once for each function it appears in

Building natively on BananaPi M2Z board (SUNxI target) against "linux-headers-5.10.16-sunxi":
user@bananapim2zero:~/aircrack-ng/rtl8812au$ make
make ARCH=arm CROSS_COMPILE= -C /usr/src/linux-headers-5.10.16-sunxi M=/home/user/aircrack-ng/rtl8812au  modules

KingKone, 2021-03-19:
same here, except I am at v5.2.20: 

```
/home/vagrant/openhd-buildroot/buildroot/output/build/openhd-rtl88xx-v5.2.20/os_dep/linux/os_intfs.c: In function ‘netdev_vir_if_close’:
/home/vagrant/openhd-buildroot/buildroot/output/build/openhd-rtl88xx-v5.2.20/os_dep/linux/os_intfs.c:2598:2: error: ‘wdev’ undeclared (first use in this function); did you mean ‘cdev’?
  wdev->iftype = NL80211_IFTYPE_MONITOR;
  ^~~~
  cdev
/home/vagrant/openhd-buildroot/buildroot/output/build/openhd-rtl88xx-v5.2.20/os_dep/linux/os_intfs.c:2598:2: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:280: recipe for target '/home/vagrant/openhd-buildroot/buildroot/output/build/openhd-rtl88xx-v5.2.20/os_dep/linux/os_intfs.o' failed

```

trying to build OpenHD on Ubuntu for NanoPi Duo2.

LinkerOK, 2021-05-22:
Use the latest branch 5.7.0

LinkerOK, 2021-05-22:
> 
> 
> Use the latest branch 5.7.0

git clone -b v5.7.0 https://github.com/aircrack-ng/rtl8812au.git

closed 2021-12-12 by kimocoder

1113870631, 2022-04-03:
> > Use the latest branch 5.7.0
> 
> git clone -b v5.7.0 https://github.com/aircrack-ng/rtl8812au.git

where is v5.7.0?

LinkerOK, 2022-05-08:
> > > Use the latest branch 5.7.0
> > 
> > 
> > git clone -b v5.7.0 https://github.com/aircrack-ng/rtl8812au.git
> 
> where is v5.7.0?

Try the latest branch - v5.6.4.2