aircrack-ng/rtl8812au #842: Garbage accompanies the transmitted packets

3brahimi, 2021-03-09:
**Issue:** Garbage accompanies the transmitted packets.
**Severity:** functionality of the driver

## Description:
Packets crafted by `libtins` (pulled today Mon 8. Mar 2021) and `scapy` are accompanied with some garbage when transmitted over. Below you can find more information about the issue I am facing.

### OS
`Linux kali 5.10.0-kali3-amd64 #1 SMP Debian 5.10.13-1kali1 (2021-02-08) x86_64 GNU/Linux`.

### WLAN Adapter
ALFA AWUS036ACH with driver version [`5.6.4.2`](https://github.com/aircrack-ng/rtl8812au/tree/v5.6.4.2) and [`5.7.0`](https://github.com/aircrack-ng/rtl8812au/tree/v5.7.0)
ALFA AWUS036H with Kali's driver works fine.

### Compiler
`g++ (Debian 10.2.1-6) 10.2.1 20210110`
`cmake version 3.18.4`

### Dependancies
`tcpdump version 5.0.0-PRE-GIT`
`libpcap version 1.11.0-PRE-GIT (with TPACKET_V3)`
`OpenSSL 1.1.1j  16 Feb 2021`
`bison (GNU Bison) 3.7.5`
`flex++ 2.6.4`

## Code to craft an open authentication packet:
`Dot11Authentication auth( ap_addr, sta_addr );`
`auth.addr3( ap_addr );`
`auth.auth_algorithm( 0 );`
`auth.auth_seq_number( 0x0001 );`
`auth.status_code( 0x0000 );`
`auto radio = RadioTap() / auth;`

## Code to send the packet:
`PacketSender sender( iface );`
`sender.send( radio );`

## Code to save the packet:
`PacketWriter w( "auth.pcap", PacketWriter::RADIOTAP );`
`w.write( radio );`

## Dumped packet in Wireshark:
`Frame 1: 60 bytes on wire (480 bits), 60 bytes captured (480 bits)`
>     Encapsulation type: IEEE 802.11 plus radiotap radio header (23)
>     Arrival Time: Mar  8, 2021 16:43:54.059643000 CET
>     [Time shift for this packet: 0.000000000 seconds]
>     Epoch Time: 1615218234.059643000 seconds
>     [Time delta from previous captured frame: 0.000000000 seconds]
>     [Time delta from previous displayed frame: 0.000000000 seconds]
>     [Time since reference or first frame: 0.000000000 seconds]
>     Frame Number: 1
>     Frame Length: 60 bytes (480 bits)
>     Capture Length: 60 bytes (480 bits)
>     [Frame is marked: False]
>     [Frame is ignored: False]
>     [Protocols in frame: radiotap:wlan_radio:wlan]
`Radiotap Header v0, Length 26`
>     Header revision: 0
>     Header pad: 0
>     Header length: 26
>     Present flags
>     MAC timestamp: 0
>     Flags: 0x10
>     Channel frequency: 2412 [BG 1]
>     Channel flags: 0x00a0, Complementary Code Keying (CCK), 2 GHz spectrum
>     Antenna signal: -50 dBm
>     Antenna: 0
>     RX flags: 0x0000
>     .... .... .... .... .... ..0. = Bad PLCP: False
`802.11 radio information`
>     PHY type: 802.11b (HR/DSSS) (4)
>     Short preamble: False
>     Channel: 1
>     Frequency: 2412MHz
>     Signal strength (dBm): -50 dBm
>     TSF timestamp: 0
`IEEE 802.11 Authentication, Flags: ........C`
>     Type/Subtype: Authentication (0x000b)
>     Frame Control Field: 0xb000
>     .000 0000 0000 0000 = Duration: 0 microseconds
>     Receiver address: AP_Manufacturer (sc:ra:mb:le:dm:ac)
>     Destination address: AP_Manufacturer (sc:ra:mb:le:dm:ac)
>     Transmitter address: STA_Manufacturer (ca:md:el:bm:ar:cs)
>     Source address: STA_Manufacturer (ca:md:el:bm:ar:cs)
>     BSS Id: 00:00:00_00:00:00 (00:00:00:00:00:00)
>     .... .... .... 0000 = Fragment number: 0
>     0000 0000 0000 .... = Sequence number: 0
>     Frame check sequence: 0x55d563c4 [unverified]
>     [FCS Status: Unverified]
`IEEE 802.11 Wireless Management`
>     Fixed parameters (6 bytes)
>>         Authentication Algorithm: Open System (0)
>>         Authentication SEQ: 0x0001
>>         Status code: Successful (0x0000)

## Transmitted packet captured by Wireshark
`Frame 18: 63 bytes on wire (504 bits), 63 bytes captured (504 bits) on interface XXXXXXX, id N`
>     Interface id: 0 (en0)
>     Encapsulation type: IEEE 802.11 plus radiotap radio header (23)
>     Arrival Time: Mar  8, 2021 17:30:21.394682000 CET
>     [Time shift for this packet: 0.000000000 seconds]
>     Epoch Time: 1615221021.394682000 seconds
>     [Time delta from previous captured frame: 903.600193000 seconds]
>     [Time delta from previous displayed frame: 903.600193000 seconds]
>     [Time since reference or first frame: 2373.833769000 seconds]
>     Frame Number: 18
>     Frame Length: 63 bytes (504 bits)
>     Capture Length: 63 bytes (504 bits)
>     [Frame is marked: False]
>     [Frame is ignored: False]
>     [Protocols in frame: radiotap:wlan_radio:wlan]
`Radiotap Header v0, Length 25'
>     Header revision: 0
>     Header pad: 0
>     Header length: 25
>     Present flags
>     MAC timestamp: 2385617569
>     Flags: 0x10
>     Data Rate: 1.0 Mb/s
>     Channel frequency: 2412 [BG 1]
>     Channel flags: 0x0480, 2 GHz spectrum, Dynamic CCK-OFDM
>     Antenna signal: -11 dBm
>     Antenna noise: -81 dBm
>     Antenna: 0
`802.11 radio information`
>     PHY type: 802.11g (ERP) (6)
>     Short preamble: False
>     Proprietary mode: None (0)
>     Data rate: 1.0 Mb/s
>     Channel: 1
>     Frequency: 2412MHz
>     Signal strength (dBm): -11 dBm
>     Noise level (dBm): -81 dBm
>     Signal/noise ratio (dB): 70 dB
>     TSF timestamp: 2385617569
>     [Duration: 496µs]
`IEEE 802.11 Authentication, Flags: ........C`
>     Type/Subtype: Authentication (0x000b)
>     Frame Control Field: 0xb000
>     .000 0001 0011 1010 = Duration: 314 microseconds
>     Receiver address: AP_Manufacturer (sc:ra:mb:le:dm:ac)
>     Destination address: AP_Manufacturer (sc:ra:mb:le:dm:ac)
>     Transmitter address: STA_Manufacturer (ca:md:el:bm:ar:cs)
>     Source address: STA_Manufacturer (ca:md:el:bm:ar:cs)
>     BSS Id: AP_Manufacturer (sc:ra:mb:le:dm:ac)
>     .... .... .... 0000 = Fragment number: 0
>     0001 0010 1100 .... = Sequence number: 300
>     Frame check sequence: 0x75a718ad [unverified]
>     [FCS Status: Unverified]
`IEEE 802.11 Wireless Management`
>    `Fixed parameters (6 bytes)`
>>        Authentication Algorithm: Open System (0)
>>        Authentication SEQ: 0x0001
>>        Status code: Successful (0x0000)
>    `Tagged parameters (4 bytes)`
>>        Tag: Diagnostic Report
>>>            Tag Number: Diagnostic Report (81)
>>>            Tag length: 115
_`[Malformed Packet: IEEE 802.11]`_
>    `[Expert Info (Error/Malformed): Malformed Packet (Exception occurred)]`
>>        [Malformed Packet (Exception occurred)]
>>        [Severity level: Error]
>>        [Group: Malformed]

j-forristal, 2022-02-11:
Not sure it it's still the case, but once upon a time Libtins would add FCS by default -- and then RTL8812 driver would add another FCS on top of that, because it didn't parse injected packet radiotap headers to determine if it already had an FCS or not.  Result was 4 extra bytes of garbage on injected frames.

Solution was to tell Libtins to not add FCS: `RadioTap().flags((RadioTap::FrameFlags)0)`

kimocoder, 2022-02-11:
I also have a script that shows issues with packet. don't have time now though