aircrack-ng/rtl8812au #845: No packet injection on ALFA AWUS1900

RodrigoCabaco, 2021-03-21:
Hello,
I have installed this driver and the 8812AU drivers for the Alfa Awus1900 
It is being detected and I am able to enable monitor mode and make a WiFi network search.
The only problem is that packet injection doesn’t seem to work.
I have tried it on 3 separate machines and I can’t make it work
A quick test on aireplay gives 0/30 on all APs
Kind regards, Rodrigo Cabaco

dinakox, 2021-03-22:
hi,
i have same problem.
0 0 acks.
ALFA AWUS 1900 realtek rtl 8814au- 5.8 driver
Linux Kali 2021 on vmware 

Best Regards,

dinakox, 2021-03-22:
hi,

please try this: https://github.com/morrownr/8814au
for me working, ACK is still  0 0, but make handshake

good luck and take care :)

RodrigoCabaco, 2021-03-24:
> hi RodrigoCabaco,
> 
> please try this: https://github.com/morrownr/8814au
> for me working, ACK is still 0 0, but make handshake
> 
> good luck and take care :)

Worked for me! 
Packet injection working 30/30 and handshake captured!
Thanks a lot for your help!

dinakox, 2021-03-24:
You're welcome

RickTorresJr, 2021-07-27:
Can confirm packet injection is working for both 2GHz and 5GHz on Alfa AWUS 1900 using this method. Thank you!

NAP3XD, 2021-07-29:
I can not my friend, still getting trying to get the driver unpacked. My make command not acting right.  Sent from Mail for Windows 10 From: Rick Torres Jr.Sent: Tuesday, July 27, 2021 12:30 PMTo: aircrack-ng/rtl8812auCc: SubscribedSubject: Re: [aircrack-ng/rtl8812au] No packet injection on ALFA AWUS1900 (#845) Can confirm packet injection is working for both 2GHz and 5GHz on Alfa AWUS 1900 using this method. Thank you!

Persico79, 2022-08-21:
Is it still working for you on kali 2022.3? Not working to me...

iDx-a, 2022-08-21:
Hi,

Yes, working without any problem