aircrack-ng/rtl8812au #850: Unable to detect channel 149 5ghz band of the hotspot provided by my samsung smartphone

mad-max-66, 2021-04-13:
I would like to connect to the 5ghz channel 149 band of the hotspot provided by my samsung smartphone but it seems that the rtl8812au driver is not enabled to connect on this channel:
`root@debian:/usr/src/rtl8812au# iwlist  wlxd0374591e180 chan
wlxd0374591e180  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Current Frequency:2.412 GHz (Channel 1)

root@debian:/usr/src/rtl8812au#`

I have tried setting various regions of use but without success.

Where is it necessary to modify the driver sources to extend the range of channels up to 5ghz 149?

Zanzaux, 2021-04-16:
Hi which driver version are you using?

mad-max-66, 2021-04-16:
The computer is an Intel Celeron dual core 64 bit. 
Linux Debian Testing kernel 5.10.0-6-amd64.
Device is  ID 2357:011f TP-Link 802.11ac WLAN Adapter (Archer T2U V3 USB Dual Band AC600).
I proceeded like this: `git clone https://github.com/aircrack-ng/rtl8812au.git && make && sudo make install`.
The installation added an 88XXau.ko kernel module.
When connecting the usb-wifi the kernel returns: `[  766.883555] usb 2-1.1: new high-speed USB device number 5 using ehci-pci
[  766.992641] usb 2-1.1: New USB device found, idVendor=2357, idProduct=011f, bcdDevice= 2.00
[  766.992650] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  766.992654] usb 2-1.1: Product: 802.11ac WLAN Adapter 
[  766.992658] usb 2-1.1: Manufacturer: Realtek 
[  766.992661] usb 2-1.1: SerialNumber: 00e04c000001
[  767.144154] usb 2-1.1: 88XXau d0:37:45:91:e1:80 hw_info[107]
[  767.214837] rtl88XXau 2-1.1:1.0 wlxd0374591e180: renamed from wlan0
[  767.795615] IPv6: ADDRCONF(NETDEV_CHANGE): wlxd0374591e180: link becomes ready`
I tried various scripts to set both the kernel module and NetworkManager the US country code which, however, even setting it manually, there is no way to connect the smartphone hotspot. 
Regarding Windows 10 it is enough to replace the driver provided by tp-link with the original Realtek rtl8812AU and you can connect without problems.
But I would very much like to be able to connect from my linux box. 
I tried to see if there was a possible option in the Makefile and try to edit the os_dep/linux /wifi_regd.c file but I have no knowledge of programming.

Your help is always welcome.
Thanks in advance.

Zanzaux, 2021-04-17:
Use version 5.6.4.2 and install using DKMS on reboot run from terminal iw reg set 00

pyro12, 2021-06-07:
I am also having this issue.  Debian Bullseye (testing), kernel 5.10.0-2-amd64

```
sudo iwlist chan wlxxxxxxx
wlxxxxxxx 32 channelsin total; available frequencies :
Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Current Frequency:2.412 GHz (Channel 1)
```

Using driver 5.6.4.2.  I've tried sudo iw reg set 00 and also editing /etc/default/crda manually (I changed to US and reverted it back to blank when I observed no change).  Many reboots.

protobits, 2021-07-23:
I think I may have a similar issue. After I upgraded to 21.04 (and then updated to latest version of the main branch of this repo) I still have some intermittent issues connecting to 5GHz networks. The behavior I see is that network manager does not usually find these networks on its own (and connects to 2.4GHz band with no issue). I can only see the 5GHz networks if I manually do "iwlist scan". After that, network manager finds the networks and auto-connects.

Now I see that `iwlist chan`  returns:
<pre>
wlxd03745e4f5a3  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Channel 149 : 5.745 GHz
          Channel 153 : 5.765 GHz
          Current Frequency=5.805 GHz
</pre>
which seems wierd since my current channel is not in the list. From `iw reg get` I get:
<pre>
global
country AR: DFS-FCC
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)

phy#1
country 99: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
	(2474 - 2494 @ 40), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5140 - 5860 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN
</pre>
(AR is correct for me). I tried "iw reg set 00" but it has no effect.

frcs6, 2021-09-11:
I have the same issue, I'm with Ubuntu 20.04.3 and when my wifi is on channel is 157, I can't connect to 5ghz.
With Windows I don't have any issue.

`iwlist chan`  :

``` console
wlx7cc2c61c4a31  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Channel 149 : 5.745 GHz
          Channel 153 : 5.765 GHz
          Current Frequency:5.22 GHz (Channel 44)
```
 
`iw reg get` :
``` console
global
country CA: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
	(5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)

phy#0
country 99: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
	(2474 - 2494 @ 40), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5140 - 5860 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN
```

frcs6, 2021-10-04:
I fix the issues with this driver : https://github.com/morrownr/8821au-20210708.git