aircrack-ng/rtl8812au #851: Hostapd clients disconnecting shortly after connecting

pinkLagoon, 2021-04-14:
Hi. I'm not sure if this is an issue with Hostapd or the rtl8812au driver, but whenever I spin up an AP using Hostapd, everything seems to work but after a client attempts to connect, it will briefly show that it was connected but then quickly drop connection.

My hostapd config is below

```
interface=wlan0
driver=nl80211
bridge=br0
ssid=TestWireless
hw_mode=g
channel=1
auth_algs=1
```

My dnsmasq config is below

```
interface=br0
dhcp-range=192.168.66.2, 192.168.66.30, 255.255.255.0, 12h
dhcp-option=3, 192.168.66.1
dhcp-option=6, 192.168.66.1
server=8.8.8.8
log-queries
log-dhcp
listen-address=127.0.0.1
```

I can successfully spin up an AP using airbase-ng. However, I really do need hostapd to work since a lot of wireless testing tools rely on hostapd rather than airbase-ng.

wkgcass, 2021-05-29:
is br0 up？if the wireless device is put into a br, hostapd will require the br to be up.

brizao, 2021-09-02:
Hi, did you solve this issue?
Because I'm also having problems with hostapd, in case I use Debian 10, but also occurs on Debian 11, after hostapd is running for some minutes, the clients disconnects and become difficult to reconnect.

I also have another usb dongle (TP Link WN722N) and never had this issue before, only with T2U Archer Plus. I think the problem is in the driver, but I don't know how to mitigate...